Moddable-OpenSource/moddable #1406: Moddable Six serial UART

mauroForlimpopoli, 2024-09-12:
Good morning,
I'm running the example 
https://github.com/Moddable-OpenSource/moddable/blob/24e3e54fd3b66379c1f91ad92969c99f941ff4cf/examples/io/serial/continuouswrite/main.js
on Moddable Six

This is the error on the xbug
C:\Users\mades\projects\moddable\examples\io\serial\continuouswrite\main.js (19) # Break: main: cannot coerce undefined to object!
How can I solve it?
How can I set a RTS and CTS port on Moddable Six ?

regards

phoddie, 2024-09-12:
You'll need to add the default Serial port to the `device` global. You can do that in the [Moddable Six](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/moddable_six/host/provider.js#L33) and [Moddable Six CDC](https://github.com/Moddable-OpenSource/moddable/blob/24e3e54fd3b66379c1f91ad92969c99f941ff4cf/build/devices/esp32/targets/moddable_six_cdc/host/provider.js#L33) host providers by adding this `Serial` section between the `I2C` and `SPI` sections:

```jsonc
	Serial: {
		default: {
			io: Serial,
			port: 1,
			receive: 44,
			transmit: 43
		}
	},
```

We'll update the repository with this change.

> How can I set a RTS and CTS port on Moddable Six ?

The `Serial` object does not currently support hardware flow control (on any device).

mauroForlimpopoli, 2024-09-13:
Hi Peter,
I did it and it's working
PS:
I change positions and it works anyway
receive: 2,
transmit: 3

Thank you

phoddie, 2024-09-13:
Thanks for confirming.

> I change positions and it works anyway
> receive: 2,
> transmit: 3

If you are using other pins, you normally would do that in your code rather than editing the device provider:

```js
new device.io.Serial({
	receive: 2,
	transmit: 3,
	port; 1,
	...
}
```

closed 2024-09-13 by phoddie