Moddable-OpenSource/moddable #1407: Error: ‘fx_SuppressedError’ undeclared here (two locations, during build of SDK)

JoshuaPK, 2024-09-15:
**Build environment:** Linux, AlmaLinux 9.4, gcc 11.4.1, cmake 3.26.5, esp-idf 5.3
**Moddable SDK version:** 5.0.0
**Target device:** ESP32 

**Description**
I followed directions in the Getting Started documentation.  I first installed esp-idf 5.3 using the method described in the Getting Started documentation ([ESP32 Install](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md)), then installed the Moddable SDK.  While attempting to do a 'make' on the SDK, I got the following compile errors:

```
/home/josh/dev/ide/moddable/build/tmp/lin/release/xsbug/mc.xs.c:19176:140: error: ‘fx_SuppressedError’ undeclared here (not in a function); did you mean ‘xsID_SuppressedError’?
19176 | 0x8b, .kind = XS_CALLBACK_X_KIND}, .value = { .callback = { fx_SuppressedError, NULL } }},
      |                                                             ^~~~~~~~~~~~~~~~~~
      |                                                             xsID_SuppressedError
make[1]: *** [/home/josh/dev/ide/moddable/build/tmp/lin/release/xsbug/makefile:362: /home/josh/dev/ide/moddable/build/tmp/lin/release/xsbug/mc.xs.c.o] Error 1

/home/josh/dev/ide/moddable/build/tmp/lin/release/xsbug/mc.xs.c:19176:140: error: ‘fx_SuppressedError’ undeclared here (not in a function); did you mean ‘xsID_SuppressedError’?
19176 | 0x8b, .kind = XS_CALLBACK_X_KIND}, .value = { .callback = { fx_SuppressedError, NULL } }},
      |                                                             ^~~~~~~~~~~~~~~~~~
      |                                                             xsID_SuppressedError
make[1]: *** [/home/josh/dev/ide/moddable/build/tmp/lin/release/xsbug/makefile:362: /home/josh/dev/ide/moddable/build/tmp/lin/release/xsbug/mc.xs.c.o] Error 1
```
I modified the affected files to change the reference of fs_SuppressedError to fx_Error and it built without problem after that.

**Steps to Reproduce**
1. Install the esp-idf toolkit as described [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md)
2. Attempt to install the Moddable SDK
3. Observe compiler error.

**Expected behavior**
A successful SDK build.

phoddie, 2024-09-15:
Thanks for the report. I'm not seeing that here on macOS, but will check on Linux.

FWIW – The error is happening when building xsbug. The `SupressedError` support is only built as part of the experimental [Explicit Resource Management proposal](https://github.com/tc39/proposal-explicit-resource-management) support. That is disabled by default. The Moddable SDK only builds it for `xst`, the XS test tool used for conformance and fuzz testing, by setting `mxExplicitResourceManagement=1` in the make file. Somehow your build of xsbug seems to have `SupressedError` partially enabled, which causing the link error.

mkellner, 2024-09-16:
Hello @JoshuaPK, is it possible that there are changes made to files in your Moddable SDK?

I've checked my `mc.xs.c` file that was created when I built the tools, searching for `Suppressed`, but I did not find it.

phoddie, 2024-09-24:
Any update here? Without some additional information to investigate, we'll close this out in the coming days. Thanks.

JoshuaPK, 2024-09-26:
This appears to be a one-off occurrence.  I retried cloning and building the Moddable repository and this problem did not re-occur.

closed 2024-09-26 by JoshuaPK

phoddie, 2024-09-26:
That's great to hear. Thanks for letting us know.