Moddable-OpenSource/moddable #1408: mcrun failed with Error `No such file or directory`

stc1988, 2024-09-21:
**Build environment:** macOS,
**Moddable SDK version:**  a9e80d0
**Target device:** esp32/m5stack, esp32/m5stack_core2, esp32/moddable_six

**Steps to Reproduce**
1. Build and install the app `examples/piu/outline/figures`  using this build command: `mcconfig -d -m -p esp32/m5stack`
2. Install the mod `CanvasPath` using `mcrun -d -m -p esp32/m5stack`

```
mcrun -d -m -p esp32/m5stack
### Error: No such file or directory
```

phoddie, 2024-09-21:
Thanks for the report. It works fine in the simulator. It looks like a side-effect of recent changes to support ESP Components. We'll take a look.

mkellner, 2024-09-22:
Hello @stc1988, I've committed a fix. Please fetch the latest and rebuild tools.

stc1988, 2024-09-22:
`mcrun` is working now, but `mcconfig` has stopped working.

```
mcconfig -d -m -p esp32/m5stack_core2
mcconfig: mod partition of size 0x40000 created from factory app partition
/Users/satoshi/.local/share/esp32/esp-idf-v5.3/tools/idf.py
/Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/figures/makefile:1526: warning: overriding commands for target `/Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/figures/ftimage.h.xsi'
/Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/figures/makefile:1457: warning: ignoring old commands for target `/Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/figures/ftimage.h.xsi'
make: *** No rule to make target `/Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/figures/xsProj-esp32/../xs_idf_deps.txt', needed by `dependencies'.  Stop.
make: *** Waiting for unfinished jobs....
rm: /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/figures/xsProj-esp32/main/idf_component.yml: No such file or directory
Using recommended ESP-IDF v5.3
```

mkellner, 2024-09-22:
My apologies. Please try again?

stc1988, 2024-09-22:
I confirmed it fixed. Thank you!!

closed 2024-09-22 by stc1988

stc1988, 2024-09-29:
@mkellner Since this issue occurs when installing Moddable using `xs-dev`, would it be possible to release version `5.0.1`?

mkellner, 2024-09-29:
I have released 5.0.1

stc1988, 2024-09-29:
Thank you for responding so quickly!