Moddable-OpenSource/moddable #1411: serial2xsbug

su-cheng-yang, 2024-09-25:
I update Moddable to the latest version, reproduced the same problem.as shown below:

Build environment:linux
Target device: esp32/m5stack
Steps to Reproduce:

Build and install the app using this build command: npm run mod --target=esp32/m5stack ./mods/look_around/manifest.json
/bin/sh: 1: [[: not found
/home/s/Projects/moddable/tools/serial2xsbug/serial2xsbug_lin.c:132: No such file or directory
make: *** [/home/s/Projects/moddable/build/tmp/esp32/debug/look_around/makefile:133: debug] Error 1

l am compling the Stack-chan firmware ,I don't know where the problem lies。, Is it  likely  Moddable environment ？

mkellner, 2024-09-25:
Hello. Maybe UPLOAD_PORT can not be found?
On my Linux machine, sometimes my device is enumerated as /dev/ttyUSB0 and sometimes /dev/ttyACM0

Maybe try `export UPLOAD_PORT=/dev/ttyUSB0` or whatever device is enumerated.
Then try again.

su-cheng-yang, 2024-09-27:
thank you for your reply

phoddie, 2026-02-04:
Closing as this appeared to have been resolved.