Moddable-OpenSource/moddable #1418: serial echo

mauroForlimpopoli, 2024-10-01:
Good morning,
I'm running the serial eco example. "https://github.com/Moddable-OpenSource/moddable/blob/public/examples/io/serial/echo/main.js"
I'm using Moddable Six and IDF 5.3
It seems the rx buffer is only 128 byte. 
Can I enlarge it?
Thank you

phoddie, 2024-10-01:
The buffer is provided by the ESP32-S3 hardware. It is fixed at 128 bytes by Espressif's design.

mauroForlimpopoli, 2024-10-02:
Hi Peter,
yes is correct, but I found this instruction on the IDF 5.3 of ESP32-S3:
uart_driver_install(UART_NUM_1, RX_BUFFER_SIZE, TX_BUFFER_SIZE, QUEUE_SIZE, &uart_queue, 0);
RX_BUFFER_SIZE should be the dimension of the buffer of RX and TX
Is it possible to add it in the sdkconfig.default ?
alternatively, how can I do to read the string of a GPS connected to a serial that is very long?
regards

phoddie, 2024-10-02:
The Serial module implementation does not use the ESP-IDF UART driver. It works directly with the hardware through the ESP-IDF HAL layer. instead In our experience, that is more reliable and has lower overhead.

> alternatively, how can I do to read the string of a GPS connected to a serial that is very long?

You can receive and concatenate the partial buffers until you find the sequence that indicates the end of the string. Assuming that the end of string is marked with a linefeed (ASCII 10), that would look something like this:

```js
new device.io.Serial({
	...device.Serial.default,
	format: "buffer",
	onReadable: function() {
		this.decoder ??= new TextDecoder;
		this.received ??= "";
		this.received += this.decoder.decode(this.read(), {stream: true});
		const linefeed = this.received.indexOf("\n");
		if (linefeed < 0) return;
		
		const result = this.received.slice(0, linefeed);
		this.received = this.received.slice(linefeed + 1);
		// use result here
	},
});
```

mauroForlimpopoli, 2024-10-02:
Hi Peter, I done the modification and I add 

trace(result, "   res\n")
trace(this.received, "   ric\n")	

this is the string 
```
###,02 Ott 2024 20:03,24Aug29164826,------------------------,
,------------------------,7,0,26,25.5, --- , --- ,
,0,0,0,0,RAMSTORE,ONE,1.1,no,fe33,
```
this is a sample of trace
```
,0,0,0,0,RAMSTO-- ,   res
```
```
,0,0,0,0,RAMSTORE,ONE,1.1,no,fe33,
3,24Aug29164826,------------------------,
,------------------------,7,0,26,25.5, --- , -7,0,26,25.5, --- , --- ,
,0,0,0,0,RAMSTORE,ONE,1.1,no,fe33,
3,24Aug29164826,------------------------,
,------------------------,   ric
```
the first characters are always not read.
how can I get the correct string?

phoddie, 2024-10-02:
Regrettably, I cannot debug a device I do not have.

If you believe the error may be in the `onReadable` code you can simply that to trace the raw results to see if the missing characters are being received:

```js
new device.io.Serial({
	...device.Serial.default,
	format: "buffer",
	onReadable() {
		try {
			trace(String.fromArrayBuffer(this.read()));
		}
		catch (e) {
			trace(e, "\n");
		}
	},
});
```

mauroForlimpopoli, 2024-10-03:
Hi Peter,
I did it, but I think the onReadable event is fired when there is a pause between two string.
Is the string is more than 128 byte the event is at the end of the string so the first part is missing
What do you think?

phoddie, 2024-10-03:
@mauroForlimpopoli, I don't understand what you are trying to communicate and you have not shared sufficient information to diagnose the situation.

Perhaps there is an issue to be addressed in the serial implementation. Or perhaps it is something else. It is impossible to say.

To have any chance of moving this forward, please provide all of the  following:

- The manufacturer and model of the GPS component you are using
- A link to the datasheet of component
- A small, standalone, complete project that demonstrates the problem under Moddable SDK v5.0.1 (or 5.0.0).

Thank you.

mauroForlimpopoli, 2024-10-04:
Hi Peter,
I tested the"echo" example putting const msg = "hello world\r\n";  with a long string (more then 200 characters). That because I wanted to simulate a string from a GPS. In this test the first characters was missing.
Today I tested the example by receiving the string from a GPS based on NEO-M8N and I receive all the strings correctly......
I'm very sorry for the mistake

phoddie, 2024-10-04:
> I tested the"echo" example putting const msg = "hello world\r\n"; with a long string (more then 200 characters). 

The `msg` is only sent once, at start-up, to indicate that the device is alive. Writing more bytes (200) than the hardware can accept (128) will result in `write` throwing an exception, so the program would terminate before echoing anything.

> Today I tested the example by receiving the string from a GPS based on NEO-M8N and I receive all the strings correctly......

Glad it is working!

closed 2024-10-04 by phoddie