Moddable-OpenSource/moddable #1419: WebSocket over SSL/TLS  fails to open

stc1988, 2024-10-02:
**Build environment:** macOS
**Moddable SDK version:** 5.0.1
**Target device:** desktop simulator

Now I try to use  [OpenAI Realtime API](https://platform.openai.com/docs/guides/realtime) with the following code, but  I got a error `/Users/satoshi/Projects/moddable/examples/io/tcp/websocketclient/websocketclient.js (59) # Break: cannot coerce undefined to object (in WebSocketClient)!`

``` js
    const ws = (this.#ws = new WebSocket({
      url: `wss://api.openai.com/v1/realtime?model=gpt-4o-realtime-preview-2024-10-01`,
      headers: new Headers([
        ["OpenAI-Beta", "realtime=v1"],
        ["Authorization", `Bearer ${apiKey}`],
      ]),
    }));
```

I think `options.dns` is set from [WebSocket.js](https://github.com/Moddable-OpenSource/moddable/blob/cb204f7cd2412ec737b24416802772dc391680f7/examples/io/tcp/websocket/WebSocket.js#L61), and that the value comes from [config.js](https://github.com/Moddable-OpenSource/moddable/blob/cb204f7cd2412ec737b24416802772dc391680f7/examples/io/tcp/websocketclient/config.js).

It doesn’t seem that `device.network.wss` is set here, but is this correct?

phoddie, 2024-10-02:
Yes, `device.network.wss` is `undefined` so the request fails because required configuration information is missing.

There is no `wss` example for 419. I've created that following the example of `https`. That will be added to this repository. For now you can [download it](https://github.com/user-attachments/files/17234777/websocketsclient.zip). This is able to establish a TLS connection to`api.openai.com`.

To use the HTML compatible `WebSocket` constructor (as your example does above), you will need to add the `wss` configuration to your project. Do that by including `$(MODDABLE)/examples/io/tcp/websocketsclient/manifest_wssclient.json`.

By the way, it is nice that issue #1419 is about ECMA-419. ;)

stc1988, 2024-10-03:
Thank you for confirming.

When I overwrite the `examples/io/tcp/websocketclient` folder and build, the following error occurred.

```
# mcrez resources
Total resource size: 5285 bytes
# cc mc.resources.c
TypeError: cannot coerce undefined to object
 at config ()
### TypeError: cannot coerce undefined to object!
make: *** [/Users/satoshi/Projects/moddable/build/tmp/mac/mc/debug/websocketclient/mc.xs.c] Error 22
```

phoddie, 2024-10-03:
> When I overwrite the examples/io/tcp/websocketclient folder and build, the following error occurred...

Yes, that is what would happen if you **overwrite** "websocketclient".

Notice that the zip file is named `websocketsclient.zip` not `websocketclient.zip` (there is an "s" between "socket" and "client", "s" for "secure"). The zip directory is intended to be placed adjacent to `websocketclient`, not replace it. That parallels the structure of the `httpclient` and `httpsclient` directories.

stc1988, 2024-10-03:
Wow! It was a basic mistake.
As a result, the RealTime API hasn't been rolled out yet, but I was able to confirm that the connection via wss is likely working.

stc1988, 2024-10-03:
I just checked, and it looks like the connection is now working 🚀

phoddie, 2024-10-03:
Good luck. Looking forward to the demo. ;)