Moddable-OpenSource/moddable #1428: `embedded:io/socket/listener` locks up Win simulator

cmidgley, 2024-10-21:
**Build environment:** Windows
**Moddable SDK version:** 5.1.0
**Target device:** Win simulator.

**Description**
I'm trying to get `embedded:network/http/server` to work on `win` simulator, but it locks up (does not respond) the simulator process and requires it to be hard terminated.  The examples for `listener/httpserver` and `listener/listen` exhibit the failure, indicating the bug is likely related to `listener`.

Both examples work on `esp32`, but for `httpserver` a trivial bug in `examples/io/tcp/websocket/WebSocket.js` on line 51 needs to be fixed to not throw an error (doesn't affect the crash issue).  Change`if (href)` to `else if (href)`, as the prior check for `object` has identified this is an attachment and there is no `href`. 

**Steps to Reproduce**
1. Run `examples/io/listener/listen` or `examples/io/listener/httpserver` on windows simulator

phoddie, 2024-10-21:
Last I checked, the unit tests for Listener passed on Windows. We'll take another look.

phoddie, 2024-10-22:
The listener on Windows has a bug where it would end up as a blocking socket, rather than non-blocking. This led to the stalls you see. There are two places to fix that., Line 176:

```c
  			ioctlsocket(tcp->skt, FIONBIO, &nonBlocking);
```

And line 577:

```c
		ioctlsocket(listener->skt, FIONBIO, &nonBlocking);
```

There is also a smaller issue – a small memory leak if an exception occurs in the constructor – which will fixed when the above changes are integrated into the SDK.

cmidgley, 2024-10-23:
Patch applied and fix confirmed.  Thank you!

closed 2024-10-23 by cmidgley