Moddable-OpenSource/moddable #1430: Win simulator with `embedded:io/socket/tcp` does not call `onError` on remove socket closure

cmidgley, 2024-10-23:
**Build environment:** Windows
**Moddable SDK version:** 5.1.0
**Target device:** Win

**Description**
When using `embedded:io/socket/tcp` the Windows simulator does not trigger `onError` when a remote socket is closed.  

**Steps to Reproduce**
1. Clone this [repo](https://github.com/cmidgley/io-socket-close)
2. `mcconfig -m -d -p win`
3. Notice that `Success!  Client-side socket was remotely closed` does not appear.

If you run this same sample on `esp32` the message correctly appears.

> Note: I saw a similar issue on legacy sockets but did not investigate it as I wanted to focus on porting to `io`.

phoddie, 2024-10-23:
Thanks for the report and the test. I converted the example to a unit test so that this will be tested on all platforms and releases moving forward. Here's what it looks like, just for reference.

<details>
<summary>onError.js</summary>

```js
/*---
description: 
flags: [async, module]
---*/

import Listener from 'embedded:io/socket/listener';
import TCP from "embedded:io/socket/tcp";

await $NETWORK.connected;

$TESTMC.timeout(5_000);

const port = 1234;

new Listener({
    port,
    onReadable() {
        const from = this.read();
        if (undefined === from)
			return $DONE("Listen failed to return socket");

        new TCP({
            from,
            onWritable() {
                this.close();
            },
            onError() {
				$DONE("incoming socket failed");
            }
        });
    },
});

new TCP({ 
    address: '127.0.0.1',
    port,
    onError() {
        $DONE();		// remote closed socket
    },
});
```

</details>

It looks like the same behavior may be present on Linux as well.

phoddie, 2024-10-24:
Here's a fix to try. It works for Linux. Maybe it will work for Windows too. ;)

In win/tcp.c, in the function `tcpTask` change:

```c
	if ((INVALID_SOCKET == tcp->skt) || tcp->done)
		return;		// closed socket
```

to:

```c
	if ((INVALID_SOCKET == tcp->skt) || tcp->done || tcp->error)
		return;		// closed socket
```

And change:

```c
else if (bytesRead < 0) {
```

to

```c
else {
```

Please let me know what you find. Thanks!

cmidgley, 2024-10-24:
Thanks.  This works great on `win` for the original test case but does not resolve the close if a write is outstanding.  If you add this before the close it will not complete the closure:

```js
this.write(ArrayBuffer.fromString('Goodbye!'));
this.close();
```

I tried to set up `testmc` to provide a unit test for this.  I got `test262` to work and run tests, but I'm doing something wrong with `testmc`.  I followed the directions several times (and did several clean builds using `mcconfig -m -d -t clean`), but I still failed with `testmc` on `win` simulator.  The symptoms are:

1) When I launch `mcconfig -m -d` on `testmc`, it crashes right away with "XS abort: unhandled exception"
2) If I enable break on exceptions, the error is `C:\moddable\tools\testmc\commodettoChecksumOut-nop.js (23) # Break: Error: ChecksumOut unimplemented!`

Also, the `Reload on abort`  is not sticky, meaning every time `mcsim` launches it resets off.  With `test262`, since it doesn't crash on load, I can it on and run tests.  I would guess that if `testmc` didn't crash, it would be the same.  I didn't know if you expect `Reload on abort` to be persistent or if it must be set on each run.

<details>
<summary>Untested: Unit test for outstanding write and close</summary>

```js
/*---
description: 
flags: [async, module]
---*/

import Listener from 'embedded:io/socket/listener';
import TCP from "embedded:io/socket/tcp";

await $NETWORK.connected;

$TESTMC.timeout(5_000);

const port = 1234;

new Listener({
    port,
    onReadable() {
        const from = this.read();
        if (undefined === from)
			return $DONE("Listen failed to return socket");

        new TCP({
            from,
            onWritable() {
                this.write(ArrayBuffer.fromString("Goodbye!"));
                this.close();
            },
            onError() {
				$DONE("incoming socket failed");
            }
        });
    },
});

new TCP({ 
    address: '127.0.0.1',
    port,
    onError() {
        $DONE();		// remote closed socket
    },
});
```
</details>

phoddie, 2024-10-25:
You have to read the pending data to get to the error. If it didn't work that way, you would potentially lose data sent immediately before a close. Updating the TCP client in the test let's it work as expected:

```js
new TCP({ 
	address: '127.0.0.1',
	port,
	onReadable() {
		this.didRead = true;
		this.read();
	},
	onError() {
		if (this.didRead)
			$DONE();
		else
			$DONE("onError before exepcted onReadable");
	},
});
```

cmidgley, 2024-10-26:
Dang that was stupid.  I know you must complete all reads to handle a close, but in my quest to make a simple test case, I over did it. Worse, I didn't test it on `esp32,` where I would have seen that my test case was flawed! Ugh. My bad.  So sorry.

But I've still been experiencing a real failure, using `io` with `HTTPServer` and `WebSocketClient`.  I finally got it to reproduce in a simple test case, using just `TCP` and `Listener`, by mirroring the same flow pattern used in those modules.  Long story short, if a socket is handed off (attached using `from` on `TCP`) and immediately closed before the other side has received the `onReadable` event, the `win` simulator fails to call `onError` for the close.  If you skip the handoff, it works fine.

The code below works on `esp32` but fails on `win`.  Interestingly, on `esp32` the `onError` event fires twice.  I've not investigated why, but on `win` it never fires and that's the issue at hand.

<details>
<summary>Test code similar to HTTP/WS handoff</summary>

```
import Listener from 'embedded:io/socket/listener';
import TCP from 'embedded:io/socket/tcp';

let httpServerSocket;
let wsServerSocket;
let messageSent = false;

// http server listening
new Listener({
    port: 1234,
    onReadable() {
        // http sever gets a connection
        trace('Listener received a client connection\n');
        const serverTCPSocket = this.read();

        // transfer to HTTP connection
        httpServerSocket = new TCP({
            from: serverTCPSocket,
            onReadable(count) {
                // got an HTTP request
                const data = this.read(count);
                trace(
                    `Received HTTP request: ${String.fromArrayBuffer(data)}\n`
                );
                this.write(ArrayBuffer.fromString('HTTP Response'));

                // transfer HTTP connection to WS connection
                trace('Transfer HTTP socket to WS server connection\n');
                wsServerSocket = new TCP({
                    from: httpServerSocket,
                    onReadable(count) {
                        const data = this.read(count);
                        trace(
                            `WS server has readable: ${String.fromArrayBuffer(
                                data
                            )}\n`
                        );
                    },
                    onWritable() {
                        trace('Closing server socket\n');
                        this.close();
                    },
                    onError() {
                        trace('WS server error\n');
                    },
                });
                trace('Closing server socket now\n');
                wsServerSocket.close();
            },
        });
    },
});

new TCP({
    address: '127.0.0.1',
    port: 1234,
    onReadable(count) {
        const data = this.read(count);
        trace(
            `Client HTTP socket received response: ${String.fromArrayBuffer(
                data
            )}\n`
        );
    },
    onWritable() {
        if (!messageSent) {
            trace('Client sending HTTP request\n');
            this.write(ArrayBuffer.fromString('HTTP Request'));
            messageSent = true;
        }
    },
    onError() {
        trace('Success!  Client-side socket closed\n');
    },
});
```
</details>

Fingers crossed I'm not sending you another stupid test case.  I've tried hard to make it clean and legit, but now my confidence is shaken.  ;-)

phoddie, 2024-10-27:
Thanks for putting together the test case. It looks a little more complicated. ;) Will give a look.

phoddie, 2024-10-27:
I'm not going to be able to run this tonight, but reading the code I think I see where there would be a problem. In win/tcp.c, modify the first few Ines as shown.

```c
void doClose(xsMachine *the, xsSlot *instance)
{
	TCP tcp = xsmcGetHostData(*instance);
	if (tcp && xsmcGetHostDataValidate(*instance, (void *)&xsTCPHooks)) {
		if (tcp->triggered)
			tcpRelease(tcp);
```

The problem is that there is an `onWritable` callback in flight – triggered from the constructor but not yet delivered. It holds a reference count. But close kills the task that would eventually deliver the callback (since no callbacks can happen after close). Since the reference count is never decrement, the socket is never closed. This change decrements the reference count if there are undelivered callbacks at the time `close()` is called.

Would you give it a try?

cmidgley, 2024-10-27:
Confirmed - it is now closing correctly for these tests.    Thank you so much!

I'm down to one last test case outstanding that is failing only on `win`.  It fills the send buffers with tons of messages followed by a close, and all I know right now is it doesn't complete the test.  I'll start working on it today and either update or close this issue based on what I learn.

cmidgley, 2024-10-27:
Well, that was an easy bug to find now that the other issues are resolved!   It's my bug (and fixed).  All my `websocket` tests using `io` now pass.

I truly appreciate the rapid, wonderful support you have provided, Peter.

closed 2024-10-27 by cmidgley

phoddie, 2024-10-28:
Glad that solves it. Thanks for the quick confirmation!

I updated the Linux and Windows implementations. macOS doesn't have this problem because it is implemented in a pretty different way. I also converted your test case to a unit test ("close-immediate.js") so that this will be consistently tested moving ahead.

Your persistence and detail in reporting these is much appreciated. Good to work out these edge cases.

Now back to reading through the giant typings PR. ;)