Moddable-OpenSource/moddable #1431: Analog Input maximum value is wrong (ESP32).

takkaO, 2024-11-07:
**Build environment:** Linux (Raspberry Pi4)
**Moddable SDK version:** 5.1.0
**Target device:** M5Stack Basic V2.61

**Description**
With analogRead, the raw value range is 0 ~ 5081.  
Since the ESP32 is a 12-bit ADC, the maximum value should be 4095.

**Steps to Reproduce**
1. Build and install the app using this build command: `mcconfig -d -m -p esp32/m5stack UPLOAD_PORT=/dev/ttyACM0`
2. Click on `Run`
3. See  Console.

**console example (input 3.3V)**
```
Resolution: 12
Analog: 5081
```

**main.js**
```js
import Timer from "timer";

debugger;

const analogInput = new device.io.Analog({
    pin: 36		// change the pin number for your development board
});

Timer.repeat(function(id) {
    trace(`Resolution: ${analogInput.resolution}\n`);
    trace(`Analog: ${analogInput.read()}\n`);
},
1000);
```

**manifest.json**
```json
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_piu.json",
		"$(MODDABLE)/modules/io/manifest.json",
		"$(MODDABLE)/examples/manifest_commodetto.json"
	],
	"modules": {
		"*": [
			"./main"
		]
	},
	"resources": {
		"*-mask": [
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-28"
		]
	}
}

```

**Expected behavior**

4095 is output at 3.3V input.

Thank you.

mkellner, 2024-11-11:
Please look for a fix in the next release.

closed 2024-11-11 by mkellner

mkellner, 2024-11-13:
Please see the [readme](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/io/analog) for new esp32 options.

takkaO, 2024-11-14:
Thank you for your swift response.