Moddable-OpenSource/moddable #1437: Touch not works on M5 Stack CoreS3

stc1988, 2024-12-15:
**Build environment:** macOS
**Moddable SDK version:** 5.3.0
**Target device:** M5 Stack CoreS3

**Description**

**Steps to Reproduce**
1. Build and install the `examples/piu/drag` app using this build command: `mcconfig -d -m  -p esp32/m5stack_cores3`

**Other information**

I think this is due to the changes in Moddable 5.2.0.

- https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack_cores3/manifest.json#L25-L26
  -  `xtouch` and `xstartupSound` seem to disable touch and startup sound.
- https://github.com/Moddable-OpenSource/moddable/blob/b8418075147c2710c402aac01510513ee886eeb9/modules/drivers/sensors/ft6206/ft6206.js#L43
  - The types of `VENDID` are fewer compared to [before the update](https://github.com/Moddable-OpenSource/moddable/blob/b8418075147c2710c402aac01510513ee886eeb9/modules/drivers/ft6206/modFt6206.c#L131).

mkellner, 2024-12-15:
My apologies.
I use as "x" in front of a config variable to disable it.
I accidentally committed.

Please remove the "x" and touch and startupSound will work.

For the second problem: we have moved that device to use ECMA-419 versions of the I<sup>2</sup>C drivers and the ft6206.js does not accept as many vendors.

Thank you for pointing it out.

Both of these fixes will be available in the next code drop.

closed 2024-12-15 by mkellner

mkellner, 2024-12-16:
I have pushed an update 5.3.1.