Moddable-OpenSource/moddable #1439: Micphone not working on M5 Stack CoreS3

mrcyan, 2024-12-17:
**Build environment:** macOS
**Moddable SDK version:** 5.3.0
**Target device:** M5 Stack CoreS3

**Description**
the example ``examples/io/audioin/capture-sync`` not working，after recording， there is no sound played.

**Steps to Reproduce**
1. Build and install the ``examples/io/audioin/capture-sync`` app using this build command: mcconfig -d -m  -p esp32/m5stack_cores3

phoddie, 2024-12-17:
Thank you for the report. We will take a look.

By the way, from your report it looks possible that the microphone is working but that it is no playing back correctly. Did you try running the `examples/io/audioin/levelmeter` example? It doesn't play the audio but logs to the console the average power level of each audio buffer.

mrcyan, 2024-12-18:
> Thank you for the report. We will take a look.
>
> By the way, from your report it looks possible that the microphone is working but that it is no playing back correctly. Did you try running the [...]

Thank you for your reply.
Yes, I tried the levelmeter example，And it ouput ``Average 30935`` continuously，I also tried the ``examples/pins/audioin/wavserver`` example， and download the wav file，It's 6 seconds, It can be played,but no sound.