Moddable-OpenSource/moddable #1440: m5paper example upload error

maskelog, 2024-12-20:
**Build environment:** Windows
**Moddable SDK version:** Current version (based on build output)
**Target device:** M5Paper (ESP32)
**Description**
The epaper-flashcards example is not working after flashing to M5Paper. The device shows no response even after hard reset.

**Steps to Reproduce**
1. Navigate to epaper-flashcards example directory:
```
PS C:\Users\xxxx\Projects\moddable\examples\piu\epaper-flashcards
```

2. Build and install the app using:
```
mcconfig -d -m -p esp32/m5paper
```

3. Flash seems successful:
- Bootloader binary size: 0x4750 bytes (36% free)
- xs_esp32.bin binary size: 0x13d7f0 bytes (69% free)
- Flash completed successfully with verification
- Hard reset performed via RTS pin

phoddie, 2024-12-21:
@maskelog – it sounds sound like the application flashed properly and your `mcconfig` build line is correct. Just to make sure the application itself is working, I ran it on the simulator with `mcconfig -d -m -p sim/m5paper`. That works just fine (I'm using macOS).

Some questions for you:

- Have you been able to run any other examples on M5Paper? examples/helloworld is always a good starting point as it has the fewest hardware dependencies.
- Have you been able to run any other examples that use the display on M5Paper? We usually start with examples/piu/balls although that will be a little slow on the ePaper display.
- Is flashcards running but not displaying? You can see this by checking the Instruments panel in xsbug. It updates once a second if all is well. You can also press the pause button to break into the debugger (or select Break from the Debug menu). If the app is running, you'll see `# Break: C: xsDebugger!` in the console.

From time to time, M5Stack changes the components in their devices. There's a small chance that the problem is due to a component change that requires a corresponding software change.

I have an M5Paper around here somewhere.... if all else fails, I can try to dig that up and give this a try.

maskelog, 2024-12-21:
@phoddie When installing the piu/balls example using the mcconfig -d -m -p esp32/m5paper command (on Windows), the Instruments Panel in xsbug displays a play/pause feature. Clicking it causes the ball to move on the M5Paper screen.

However, in the epaper-flashcards example, the play/pause feature does not appear.

It seems that some examples work correctly while others do not due to this issue.