Moddable-OpenSource/moddable #1443: Build environment variables overwritten by nested manifests

HipsterBrown, 2024-12-26:
**Build environment:** macOS
**Moddable SDK version:** 5.3..0
**Target device:**  NodeMCU ESP8266

**Description**
When targeting the ESP8266, the manifest for servo support includes a build environment variable for `ARDUINO_ESP8266` that is expected to be in `$(USERHOME)/esp/esp8266-2.3.0/`. This conflicts with the conventional setup location for xs-dev: `$(USERHOME)/.local/share/esp/esp8266-2.3.0/` (or `$(USERHOME)/xs-dev/esp/esp8266-2.3.0/` on Windows). Setting the `ARDUINO_ESP8266` build variable in my project manifest fails to fix the issue due to the [`parseBuild` method](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/mcmanifest.js#L2171) in `mcmanifest.js` setting the environment variable, whether or not it has been set before. 

**Steps to Reproduce**
1. `xs-dev setup --device esp8266`
2. `xs-dev run --example pins/servo --device esp8266/nodemcu`
3. See error: `### Error: '$(ARDUINO_ESP8266)/libraries/Servo/src/esp8266/*': directory not found!`

**Expected behavior**
Setting the build environment variable should resolve the module paths correctly. 

**Other information**
Fixing this might resolve in breaking changes if other parts of the build system rely on this behavior,

phoddie, 2025-01-02:
Your changes look reasonable. The key point is the one you make above – this could break other parts of the build.

We use the `build` section of the manifest primarily as a shortcut for local paths within the Moddable SDK - `$(PIU)`, `$(MODULES)`, `$(XS)`, etc. Those generally don't want to be overridden by external environment variables, although perhaps it could be useful in some cases. The risk is that it increases the chances of an unexpected interaction with the build environment, especially for general names like `MODULES`.

Another approach would be to allow the manifest to control the behavior. That would keep the current behavior by default but allow this case to work.

HipsterBrown, 2025-01-03:
> Another approach would be to allow the manifest to control the behavior. That would keep the current behavior by default but allow this case to work.

Would this be a new field or option for the manifest, like "overrides"? I also noticed a different PR for "redirects" that could be related?

phoddie, 2025-01-04:
I don't think we should change the existing manifest behavior.  Adding new features to the manifest is probably more than is needed here.

The servo manifest wants to refer to files in the Arduino repo that is used in the build. If this was for ESP32, it would just use `$(IDF_PATH)`. I think the solution is to ensure that `ESP_BASE` and `ARDUINO_ROOT` are always defined when manifests are processed. This requires moving a little logic from the ESP8266 makefile to `mcconfig`. Try adding the following code at the top of the `run()` method in `tools/mcconfig.js` (remember to rebuild tools after making the change):

```js
		if (this.platform === "esp") {
			let base = this.environment.ESP_BASE ?? this.getenv("ESP_BASE");
			if (!base) {
				base = this.resolveDirectoryPath(this.getenv("HOME") + "/esp/");
				if (!base)
					throw new Error("Arduino SDK not found");
			}
			this.environment.ESP_BASE = base;
			this.setenv("ESP_BASE", base);
		
			this.environment.ARDUINO_ROOT = base + "/esp8266-2.3.0";
			this.setenv("ARDUINO_ROOT", base);
		}
```

Then `pins/servo/manifest.json` can remove the `build` section and use `ARDUINO_ROOT` instead of `ARDUINO_ESP8266` :

```jsonc
{
	"platforms":{
		"esp": {
			"modules": {
				"pins/servo": "$(MODULES)/pins/servo/servo",
				"*": [
					"$(MODULES)/pins/servo/esp/*",
					"$(ARDUINO_ROOT)/libraries/Servo/src/esp8266/*",
					"$(ARDUINO_ROOT)/libraries/Servo/src/*"
				]
			},

```

This works cleanly in my testing. Let me know if this works for you with xs-dev too. If so, it will need some adjustments for Windows before committing.

HipsterBrown, 2025-01-05:
This works for me as well. I wonder where these types of environment variable requirements should be documented for folks using the Moddable SDK directly. 

---

While we're looking into these build variables, I noticed a similar issue in the `build/devices/pico/manifest.json`. It defines `PICO_EXTRAS_DIR` and `PICO_EXAMPLES_DIR` in the `build` field. xs-dev automatically defines these environment variables in the `xs-dev-export.sh`. I supposed this would require another conditional block for the `pico` devices in `run()` method of `tools/mcconfig.js`.

phoddie, 2025-01-06:
Thanks for confirming the changes work. I'll work on finishing those up for Windows and committing.

--- 

The build variables, beyond `$(MODDABLE)` are platform dependent and should generally be explained in the document for that platform. For example, the `$(IDF_PATH)` environment variable is explained in the [ESP32 Install instructions](https://www.moddable.com/documentation/devices/esp32#mac-instructions). `$(ARDUINO_ROOT)` is not explained in the [ESP8266 document](https://www.moddable.com/documentation/devices/esp8266), so that should be updated.

Regarding Pico, what you describe sounds correct. As currently written, `$(PICO_EXAMPLES_DIR)` and `$(PICO_EXTRAS_DIR)` are broken for use by manifests if they are not in the default location. At the moment, no manifests use either, so it is problem waiting to happen...

HipsterBrown, 2025-01-06:
> Regarding Pico, what you describe sounds correct. As currently written, $(PICO_EXAMPLES_DIR) and $(PICO_EXTRAS_DIR) are broken for use by manifests if they are not in the default location. At the moment, no manifests use either, so it is problem waiting to happen...

I hit this issue a couple weeks ago while working on a quick project 😅

phoddie, 2025-01-22:
The changes discussed above are part of Moddable SDK 5.4.0.

closed 2025-01-22 by phoddie