Moddable-OpenSource/moddable #1448: camera-server-motion-jpeg does not work on Moddable SDK 5.3.3

kitazaki, 2025-01-04:
**Build environment:** macOS
**Moddable SDK version:** 5.3.3
**Target device:** ESP32S3 (XIAO ESP32S3 Sense and M5Atom S3R M12)

**Description**
camera function added on Moddable SDK 5.2.0.
It works on 5.2.0 and 5.3.0, but does not work on 5.3.3

**Steps to Reproduce**
build sample program "camera-server-motion-jpeg" and run.

| Moddable SDK Version | XIAO ESP32S3 Sense | M5Atom S3R M12
| ------------- | ------------- | ------------- |
| 5.2.0 | OK | OK |
| 5.3.0 | OK | OK |
| 5.3.3 | NG | NG |

**Expected behavior**
Motion JPEG data comes from a device after accessed to the URL, but no data comes with HTTP response headers and HTTP connection is closed.

* OK
[HTTP Request]
GET /camera HTTP/1.1
Host: 192.168.3.157:8080
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:133.0) Gecko/20100101 Firefox/133.0
Accept: image/avif,image/webp,image/png,image/svg+xml,image/*;q=0.8,*/*;q=0.5
Accept-Language: ja,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Connection: keep-alive
Referer: http://192.168.3.157:8080/
Priority: u=4, i
[HTTP Response]
HTTP/1.1 200 FILL_THIS_IN
content-type: multipart/x-mixed-replace; boundary="419cameraboundary"
--419cameraboundary
content-length: 5356
content-type: image/jpeg
--419cameraboundary
content-length: 5356
content-type: image/jpeg
(continues)

* NG
[HTTP Request]
GET /camera HTTP/1.1
Host: 192.168.3.157:8080
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:133.0) Gecko/20100101 Firefox/133.0
Accept: image/avif,image/webp,image/png,image/svg+xml,image/*;q=0.8,*/*;q=0.5
Accept-Language: ja,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Connection: keep-alive
Referer: http://192.168.3.157:8080/
Priority: u=4, i
[HTTP Response]
HTTP/1.1 200 FILL_THIS_IN
content-type: multipart/x-mixed-replace; boundary="419cameraboundary"

**Images**

* OK
[image]

* NG
[image]

**Other information**
no information and error on xsbug LOG.

Wi-Fi connected to "SSID"
IP address 192.168.3.157
Camera ready.
Web page: http://192.168.3.157:8080
Camera Motion JPEG camera: http://192.168.3.157:8080/camera

phoddie, 2025-01-04:
@kitazaki – Thank you for reporting. Not much changed with the camera between 5.3.0 and 5.3.3 so maybe this is related to the networking changes. We'll take a look.

phoddie, 2025-01-05:
It looks like the problem was caused by a change the HTTP server. In the case where the response body length is unknown (no content-length and not a chunked response), the connection was being closed immediately after sending the headers. It works for me by replacing [this line](https://github.com/Moddable-OpenSource/moddable/blob/9198d0111edab82bf68bd4e4747add00df0c3039/examples/io/listener/httpserver/httpserver.js#L316) with:

```js
					if (0 === this.#remaining) {
```

Would you confirm that also solves the problem for you? Thank you.

kitazaki, 2025-01-05:
@phoddie 
Thank you, It works fine for me.

closed 2025-01-05 by kitazaki

phoddie, 2025-01-05:
Excellent! Thank you. We'll get that merged for the next update.