Moddable-OpenSource/moddable #1450: The characteristic class in BLE has a property `properties`.

stc1988, 2025-01-13:
When I was checking the data contained in the characteristic, I found a property `properties`, which is of the Number type.
This is not documented, but is it the same as what the GATT service has?

phoddie, 2025-01-13:
Yes, I think you are correct. The object passed to the `onCharacteristic` callback contains a `properties` property which corresponds to the GATT Characteristic Properties. It looks like we don't use that in the Moddable SDK examples.

Would you like to update the documentation or would you prefer that I do that?

phoddie, 2025-01-15:
The properties are a comma separated list in the JSON service description. Here they are flags, as you surmise. The definition of values comes from the BLE standard. Here is the relevant section of the ESP-IDF header `ble_gatt.h` that defines them:

```c
/**
 * @defgroup ble_gatt_chr_properties Generic Attribute Profile (GATT) Characteristic Properties
 * @{
 */

/** Characteristic property: Broadcast. */
#define BLE_GATT_CHR_PROP_BROADCAST                     0x01

/** Characteristic property: Read. */
#define BLE_GATT_CHR_PROP_READ                          0x02

/** Characteristic property: Write Without Response. */
#define BLE_GATT_CHR_PROP_WRITE_NO_RSP                  0x04

/** Characteristic property: Write. */
#define BLE_GATT_CHR_PROP_WRITE                         0x08

/** Characteristic property: Notify. */
#define BLE_GATT_CHR_PROP_NOTIFY                        0x10

/** Characteristic property: Indicate. */
#define BLE_GATT_CHR_PROP_INDICATE                      0x20

/** Characteristic property: Authenticated Signed Write. */
#define BLE_GATT_CHR_PROP_AUTH_SIGN_WRITE               0x40

/** Characteristic property: Extended Properties. */
#define BLE_GATT_CHR_PROP_EXTENDED                      0x80
```

It appears that only READ, WRITE_NO_RSP, WRITE, NOTIFY, INDICATE and EXTENDED are currently used.

For the documentation, I suggest adding `properties` to the table in Class Characteristic > Properties:

| Name | Type | Description |
| --- | --- | :--- |
| `...` | `...` | ...
| `properties` | `number` | Bit field of GATT Characteristic Properties (2 `read` = 2, `writeNoResponse` = 4, `write` = 8, `notify` = 16, `indicate` = 32, `extended` = 128)

What do you think?

stc1988, 2025-01-16:
Thank you for your polite explanation. I think the content you suggested is good.

phoddie, 2025-01-16:
Great, thank you. I've merged the change for our (overdue!) January Moddable SDK release.

closed 2025-01-16 by phoddie