Moddable-OpenSource/moddable #1453: moddabl six "pins/audioout" not found!

mauroForlimpopoli, 2025-01-17:
Good morning, 
I'm trying to import part of the example "examples/piu/sound" in my project.
I only need to play a sound in the internal speaker
I added to my main
```
import Sound from "piu/Sound"
import {} from "piu/MC"; 
Sound.volume = 0.5;
let sound = new Sound({
    path: "bflatmajor.wav"
});
sound.play();
```
and this is my manifest
```
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_piu.json",
		"$(MODULES)/input/keyboard/manifest.json",
		"$(MODDABLE)/modules/files/file/manifest.json",
		"$(MODULES)/network/wifi/manifest.json",
		"$(MODDABLE)/examples/manifest_net.json",
		"$(MODDABLE)/modules/network/ble/manifest_server.json",
		"$(MODDABLE)/modules/io/manifest.json",
		"$(MODDABLE)/modules/network/ble/manifest_client.json",
		"$(MODDABLE)/modules/drivers/peripherals/ds3231/manifest.json",
		"$(MODDABLE)/modules/drivers/rtc/manifest.json",
		"$(MODULES)/pins/i2c/manifest.json",
		"$(MODULES)/network/http/manifest.json",
		"$(MODDABLE)/modules/files/preference/manifest.json",
		"$(MODDABLE)/modules/files/flash/manifest.json",
		"$(MODULES)/drivers/onewire/manifest.json"
	],
	"creation":{
		"stack":700
	},
	"modules": {
		"pins/smbus": "$(MODULES)/pins/smbus/smbus",
		"*": [
			"./main",
			"./assets",
			"$(MODULES)/input/keyboard/*",
			"$(MODULES)/files/flash/*",			
			"$(MODULES)/network/http/*",
			"$(MODULES)/network/wifi/*",
			"./funcSleep",
			"$(MODULES)/network/sntp/*",
			"$(MODULES)/drivers/neopixel/*",
			"$(MODULES)/drivers/neopixel/esp32/*",
			"$(MODULES)/drivers/rtc/rtc",
			"$(MODULES)/drivers/rtc/ds3231"
		],
		"piu/Sound": [
			"$(MODULES)/piu/MC/piuSound"
		]
	},
	"resources":{
		"*-alpha": [
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-18",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Regular-20",
			"$(MODDABLE)/examples/assets/fonts/arial20",
			"$(MODDABLE)/examples/assets/fonts/arial32",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-28",
			"$(MODDABLE)/examples/assets/fonts/impact64",
			"$(MODDABLE)/examples/assets/fonts/futura"
		],
		"*-image(50)": [
			"$(MODDABLE)/examples/assets/images/Bluetooth-icon",
			"$(MODDABLE)/examples/assets/images/Wifi-icon",
			"$(MODDABLE)/examples/assets/images/unlock-icon",
			"$(MODDABLE)/examples/assets/images/lock-icon",
			"$(MODDABLE)/examples/assets/images/down-icon",
			"$(MODDABLE)/examples/assets/images/memory-icon",
			"$(MODDABLE)/examples/assets/images/sun-icon",
			"$(MODDABLE)/examples/assets/images/moon-icon",
			"$(MODDABLE)/examples/assets/images/rs232-icon",
			"./assets/logo",
			"./assets/qrcode_google"
		],
		"*-mask": [
			"$(MODDABLE)/examples/assets/images/thermometer-mask",
			"$(MODDABLE)/examples/assets/images/battery-mask",
			"$(MODDABLE)/examples/assets/images/switch-mask"	,
			"$(MODDABLE)/examples/assets/images/option-mask"	,
			
			"./assets/lock-icon-mask",
			"./assets/unlock-icon-mask",
			"./assets/down-icon-mask",
			"./assets/memory-icon-mask",
			"./assets/moon-icon-mask",
			"./assets/rs232-icon-mask",
			"./assets/sun-icon-mask",
			"./assets/battery-icon-mask",
			"./assets/Bluetooth-icon-mask",
			"./assets/Wifi-icon-mask",
			"./assets/gps-icon-mask",
			"./assets/led_on-disc-mask",
			"./assets/switch_off-mask",
			"./assets/switch_on-glyph-mask",
			"./assets/attendere",
			"./assets/print",
			"./assets/downCloud",
			"./assets/one100",
			"./assets/one100E"
		],			
		"*": [
			"./assets/print-icon",
			"./assets/sms2-icon",
			"./assets/opts-icon",
			"./assets/menuIcon",
			"$(MODDABLE)/examples/assets/sounds/bflatmajor"
		]
	},	
	"preload": [
		"neopixel",
		"http",
		"sntp",
		"file",
		"ble",
		"preference",
		"flash",
		"keyboard",
		"time",
		"timer",
		"funcSleep",
		"wifi",
		"keyboard",
		"assets",
		"ds1307",
		"rtc",
		"smbus",
		"DS18X20"
	],
	"platforms": {
		"esp": {
			"config": {
				"rotation": 90
			}
		},
		"esp32": {
			"config": {
				"rotation": 90,
				"brightness": "off",
				"file": {
					"root": "/mod/"
				},
				"onewire": {
					"pin": "45"
				}
			},
			"defines": {
				"ls013b4dn04": {
					"dither": 1,
					"updateAll": 1
				}
			},
			"include": [
				"$(MODULES)/pins/digital/manifest.json"
			],
			"modules": {
				"*": [
				"$(MODULES)/files/flash/*",
				"$(MODULES)/files/flash/esp32/*"
				]
			},						
			"build": {
				"PARTITIONS_FILE": "./sdkconfig/partitions.csv"
			},
			"modules": {
				"ota": "$(BUILD)/devices/esp32/modules/ota/*"
			},
			"preload": "ota" 			
		}
    },
	"ble":{
		"*": [
			"./bleservices/*",
			"./bleservicesCli/*"
		]
	}	
}

```
The example works correctly in the moddable six, but in my project I get:
```
ReferenceError: module "pins/audioout" not found
### ReferenceError: module "pins/audioout" not found!
```
What I wrong ?

phoddie, 2025-01-18:
> I'm trying to import part of the example "examples/piu/sound" in my project.

You have included the piuSound module, like the example. That's good. But, you did not [include the pins/i2s module](https://github.com/Moddable-OpenSource/moddable/blob/9198d0111edab82bf68bd4e4747add00df0c3039/examples/piu/sound/manifest.json#L5) that the example does. That contains the audioout driver.

mauroForlimpopoli, 2025-01-20:
It's works. Thank you Peter, sorry for the oversight.

closed 2025-01-20 by phoddie