Moddable-OpenSource/moddable #1465: Function to string does not comply with ECMAScript specification

ayuan0828, 2025-03-02:
### Description

```markdown
When a user-defined function (not a built-in function) is converted to a string, the XS engine returns a placeholder for '[native code]' instead of the actual function body string. This is inconsistent with the ECMAScript specification and mainstream engine behavior, and may have an impact on the relevant code.
```

### Version

```
a29e8f5
```

### Testcase

```
var functionArray = [
    function () {}
];
functionArray = functionArray + functionArray;
print(functionArray.length);
```

### Output

```
62
```

### Expected output

```
28
```

phoddie, 2025-03-02:
Thank you for your report.

Different does not always mean incorrect. XS is in conformance with the ECMA-262 specification on this matter. As noted in the ECMA-262 notes on ECMAScript 2019:

> ECMAScript 2019... clarif(ied) `Function.prototype.toString` by requiring that it either return the corresponding original source text or a standard placeholder What you see returned by XS is the "standard placeholder." 

The specification for [`Function.prototype.toString`](https://tc39.es/ecma262/#sec-function.prototype.tostring) explains that the string returned depends on the value returned by the [`HostHasSourceTextAvailable`](https://tc39.es/ecma262/#sec-hosthassourcetextavailable)  host function, which may return `false` to allow "host environments to prevent the source text from being provided." On the web, this mechanism is used to deny access to source code in certain secure contexts; in XS, to preserve limited embedded storage space and RAM by not having to store the source text.

----

If I may ask, you seem to be doing some automated comparison across engines. That's certainly interesting. Would you share more about your testing approach? Thank you.

ayuan0828, 2025-03-03:
Yes, I'm doing fuzziness testing, which can output some differences between engines, but the gap between difference and incorrectness is hard to tell apart. I'm sorry if I've caused you any trouble.

phoddie, 2025-03-03:
> I'm doing fuzziness testing, which can output some differences between engines...

That's very cool. It should be able to detect subtle conformance issues in implementations. 

> ...but the gap between difference and incorrectness is hard to tell apart

Yes, I imagine. For example,  error strings are completely different across engines. And, as we have seen, ECMA-262 leaves some room for implementations to differ. What approach are you taking to distinguishing problematic differences from non-confirming behaviors?

closed 2025-03-03 by phoddie

ayuan0828, 2025-03-04:
Typically, automatic program filtering is used for initial distinction. If items are not filtered out, the content of ECMA-262 is analyzed to make a judgment.