Moddable-OpenSource/moddable #1466: conversationalAI  app Error: display list overflowed (in onIdle)!

stc1988, 2025-03-06:
**Build environment:** macOS
**Moddable SDK version:** Moddable5.5
**Target device:**  desktop simulator

**Steps to Reproduce**

1. I asked the AI to tell me about “Moddable SDK”.
2. The app crashed while the AI was conversing.

[image]

```
/Users/satoshi/Projects/moddable/modules/piu/MC/piuView.c (1124) # Break: Error: display list overflowed (in onIdle)!
```

phoddie, 2025-03-06:
Hey, thanks for trying it out. Nice description of the Moddable SDK you have there. ;)

The app terminates because of an unhandled exception thrown when the display list overflow is reported. You can simply enlarge the display list in main.js.

https://github.com/Moddable-OpenSource/moddable/blob/b11c527e1e2fac50deea2e375ddaf5995f670e9c/contributed/conversationalAI/main.js#L49

The same error does not happen on the device, however. The reason is that display list entries are bigger on the simulator because pointers are 64-bit on computers and only 32-bit on our usual target embedded devices. To compensate for that, we're going to make a change to automatically increase the size of the requested display list length when compiled for 64-bit point. That will allow for more consistent results between devices and the simulator. You can try that out. After retrieving the `displayListLength` here...

https://github.com/Moddable-OpenSource/moddable/blob/b11c527e1e2fac50deea2e375ddaf5995f670e9c/modules/commodetto/commodettoPocoCore.c#L95

... add these lines:

```c
	if (sizeof(void *) > 4)
		displayListLength += displayListLength >> 1;		// compensate for bigger pointers on 64-bit systems
```

stc1988, 2025-03-09:
@phoddie 

Thank you for checking it so quickly.

> You can try that out. After retrieving the displayListLength here...
> 
> [moddable/modules/commodetto/commodettoPocoCore.c](https://github.com/Moddable-OpenSource/moddable/blob/b11c527e1e2fac50deea2e375ddaf5995f670e9c/modules/commodetto/commodettoPocoCore.c#L95)
> 
> Line 95 in [b11c527](https://github.com/Moddable-OpenSource/moddable/commit/b11c527e1e2fac50deea2e375ddaf5995f670e9c)
> 
>  int displayListLength = xsmcToInteger(xsArg(4)); 
> ... add these lines:
> 
> 	if (sizeof(void *) > 4)
> 		displayListLength += displayListLength >> 1;		// compensate for bigger pointers on 64-bit systems

This code worked as expected.

closed 2025-03-10 by phoddie