Moddable-OpenSource/moddable #1467: conversationalAI app crash and reboot  after After the AI agent finished speaking.

stc1988, 2025-03-09:
**Build environment:** macOS,
**Moddable SDK version:** 5.5
**Target device:** M5stack Core2

**Steps to Reproduce**
1. I talk to the AI about anything.
2. App crash and reboot after After the AI agent finished speaking.

**Other information**

I was able to reproduce the issue with the instrumented version and confirmed the following error.

 ```
W (29623) i2s_platform: i2s controller 0 has been occupied by i2s_driver
E (29623) i2s_common: i2s_new_channel(968): no available channel found
i2s_new_channel failed
E (29636) i2s_pdm: i2s_channel_init_pdm_rx_mode(453): input parameter 'handle' is NULL
E (29636) i2s_common: i2s_del_channel(1024): input parameter 'handle' is NULL
i2s_channel_init_pdm_rx_mode failed
E (29652) i2s_common: i2s_channel_enable(1156): input parameter 'handle' is NULL
E (29652) i2s_common: i2s_del_channel(1024): input parameter 'handle' is NULL
i2s_channel_enable failed
instruments: 310,10,0,0,1,2,0,120,516,0,80,2390343,7,2,47384,262144,63584,65520,1152,8192,0,3,1,0,69,0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40117812  PS      : 0x00060033  A0      : 0x80086301  A1      : 0x3ffb2470  
--- 0x40117812: playedBuffer at /Users/satoshi/Projects/moddable/modules/io/audioout/esp32/audioout-esp32.c:543

A2      : 0x3f9db110  A3      : 0x3ffb2490  A4      : 0x3ffdb324  A5      : 0x3ffb2480  
A6      : 0x3ffb0178  A7      : 0x3ffb4640  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x3ffb9534  A11     : 0x3ffb9534  A12     : 0x8008c886  A13     : 0x3ffb2470  
A14     : 0x3ffdb324  A15     : 0x00060023  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000015  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4011780f:0x3ffb2470 0x400862fe:0x3ffb2490 0x40082089:0x3ffb24d0 0x40083e8d:0x3ffb24f0 0x40089f73:0x3ffbbdf0 0x400d2652:0x3ffbbe10 0x4008d767:0x3ffbbe30 0x4008c649:0x3ffbbe50
--- 0x4011780f: playedBuffer at /Users/satoshi/Projects/moddable/modules/io/audioout/esp32/audioout-esp32.c:539
0x400862fe: i2s_dma_tx_callback at /Users/satoshi/.local/share/esp32/esp-idf/components/esp_driver_i2s/i2s_common.c:724
0x40082089: shared_intr_isr at /Users/satoshi/.local/share/esp32/esp-idf/components/esp_hw_support/intr_alloc.c:445
0x40083e8d: _xt_lowint1 at /Users/satoshi/.local/share/esp32/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x40089f73: xt_utils_wait_for_intr at /Users/satoshi/.local/share/esp32/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /Users/satoshi/.local/share/esp32/esp-idf/components/esp_hw_support/cpu.c:55
0x400d2652: esp_vApplicationIdleHook at /Users/satoshi/.local/share/esp32/esp-idf/components/esp_system/freertos_hooks.c:58
0x4008d767: prvIdleTask at /Users/satoshi/.local/share/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)
0x4008c649: vPortTaskWrapper at /Users/satoshi/.local/share/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

phoddie, 2025-03-10:
I don't have an M5Stack CoreS2 to test, but there is a change you might try. Try changing this:

https://github.com/Moddable-OpenSource/moddable/blob/b11c527e1e2fac50deea2e375ddaf5995f670e9c/modules/io/audioout/esp32/audioout-esp32.c#L115-L122

...to this:

```c
	if (audioOut->tx_handle) {
		i2s_channel_disable(audioOut->tx_handle);
		i2s_del_channel(audioOut->tx_handle);
		audioOut->tx_handle = NULL;
	}
#ifdef MODDEF_AUDIOOUT_I2S_PDM_PIN
	gPDMAudioOutBusy = 0;
#endif
```

stc1988, 2025-03-10:
@phoddie 
The changes to the code made it work as expected!

phoddie, 2025-03-10:
Nice. Thanks for the help! I'll get that committed

closed 2025-03-10 by phoddie