Moddable-OpenSource/moddable #1468: Variables that survive soft reset

mauroForlimpopoli, 2025-03-13:
I worked with Moddable Two and IDF 4.2 and used 
```
RTC_NOINIT_ATTR int buff[20];
```
in a setting.c file to preserve the variables during a soft reset or crash.
I need to write them every 10 seconds so I can't use NVS 
Now on Moddable Six and IDF 5.3.0 this command is available, but it seems doesn't work.
Can you help me? have you other ideas ?
regards

phoddie, 2025-03-13:
My recollection is that RTC memory on ESP32 is only retained across sleep (deep and light). Any crash, reset, restart clears RTC memory.

> Now on Moddable Six and IDF 5.3.0 this command is available, but it seems doesn't work.

What "command" are you referring to?

If you can provide a working example of what you want to do using the ESP-IDF, I can try to help you integrate that into the Moddable SDK.

mauroForlimpopoli, 2025-03-13:
in my main.js 
```
function saveDataString() @ "xs_saveDataString"
function readDataString() @ "xs_readDataString"
saveDataString(0,5); // put value 5 in the buff[0]
readDataString(0);
```
this is in the setting.c
```
#include "xsmc.h"
#include "stdint.h"
#include "string.h"
#include "mc.xs.h"			// for xsID_ values
#include  "xs.h"
........
RTC_NOINIT_ATTR int buff[20];  /// must survive to soft restar of crash
RTC_DATA_ATTR int puntLoc = 0;
RTC_DATA_ATTR int value1 = 0;

void xs_saveDataString(xsMachine * the)
{
	puntLoc = xsmcToInteger(xsArg(0));
	value1 = xsmcToInteger(xsArg(1));
	buff[puntLoc] = value1;
}
void xs_readDataString(xsMachine * the)
{
	int locBuff = xsmcToInteger(xsArg(0));
	int greeting = buff[locBuff];
	xsResult = xsInteger(greeting);
}
```
this works in the idf 4.2 but not in the 5.3.0, 
for "command" i mean that the attibute for variable 
RTC_NOINIT_ATTR 
RTC_DATA_ATTR
are valid in the idf 5.3.0

phoddie, 2025-03-13:
I see. So changing the ESP-IDF caused you a probelm. That seems like an issue for Espressif.

closed 2025-03-13 by phoddie