Moddable-OpenSource/moddable #1470: [ChatAudioIO]this.error is undefined in onStateChanged callback

stc1988, 2025-03-17:
**Build environment:** macOS
**Moddable SDK version:** 5.5.0
**Target device:** desktop simulator, 

When onStateChanged is triggered with the "FAILED" state, referencing this.error results in an error. It seems that this does not refer to the expected instance inside the callback function.

**Code to reproduce:**

``` js
const chat = new ChatAudioIO({
  ...
  onStateChanged: (state) => {
    switch (state) {
      case "FAILED":
        trace(`${this.error}\n}`)
        break;
    }
  }
});
```

phoddie, 2025-03-17:
Thank you for the report. Please provide some additional detail:

1. What is the error that is results from referencing `this.error`?
2. How can we reproduce this behavior?

stc1988, 2025-03-17:
[This ia](https://github.com/stc1988/moddable-labs/blob/main/examples/app/conversationalAI/main.js) sample app I made.

To reproduce this behavior, just set invalid api_token.
`mcconfig -d -m -p sim/moddable_six openAIKey=$OPENAI_API_KEYS`

These are console log.
```
[onStateChanged]undefined => CONNECTING
[onStateChanged]CONNECTING => FAILED
/Users/satoshi/Projects/moddable-labs/examples/app/conversationalAI/main.js (35) # Break: TypeError: cannot coerce undefined to object (in onStateChanged)!
```

stc1988, 2025-03-17:
Modify [this line](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/services/chatAudioIO/ChatAudioIO.js#L65) to below code works as I expected, but I don’t know if this is the appropriate way.

``` js
this.onStateChanged = options.onStateChanged.bind(this) ?? callback;
```

phoddie, 2025-03-17:
Great, thanks. We'll take a look.

phoddie, 2025-03-18:
The `onStateChanged` function is an arrow function, so it is bound to `this` of the surrounding code. That

```js
const chat = new ChatAudioIO({
  ...
  onStateChanged: (state) => {
```

Because `this` is not the `chat` instance, you cannot use it to access `error` of the chat instance. Instead, you can access `chat` directly:

```js
      case "FAILED”: 
	 trace(`${chat.error}\n}`) 
	 break;
```

Arrow functions are powerful, but also a little but confusing. ;)

stc1988, 2025-03-18:
```
      case "FAILED”: 
	 trace(`${chat.error}\n}`) 
	 break;
```

Sorry, I overlooked the basic part.
The issue has been resolved, so I will close it.

closed 2025-03-18 by stc1988

phoddie, 2025-03-18:
No problem. ;)