Moddable-OpenSource/moddable #1471: Moddable Two/Six touch screen

mauroForlimpopoli, 2025-03-18:
I'm using Moddable Two with add a real time clock DS3231 on the I2C bus.
Sometimes when touch the screen the board resets without reason. No error from the debugger.
I'm thinking to a collision with the 2  devices I2C. ?!?!?
From the feature of the Moddable Two it seems the touch screen doesn't use interrupt line but the touch is continually searched
Is it possible to stop the searching of touch and to active again ?
How can I avoid the collision between the peripherals?
In the Moddable Six the touch screen works on interrupt, and the problem now is not present, but can I do the same thing?

At the end, sometimes for various reason I get the error
 "throw new Error("in use");" .... or other texts
or
"xsUnknownError("in use");" .... or other texts
and the device restarts
Is it possible to avoid this? I mean only to receive a code of error without restar

regards

mauroForlimpopoli, 2025-03-18:
PS: maybe is possible just only with 
 "active: false" ....of the Container,Applicaton,Scroller....

phoddie, 2025-03-19:
The two I²C devices shouldn't be accessed at the same time. If you are using a worker, then maybe there is a bug allowing it.

> At the end, sometimes for various reason I get the error
> "throw new Error("in use");" .... or other texts
> or
> "xsUnknownError("in use");" .... or other texts
> and the device restarts
> Is it possible to avoid this? I mean only to receive a code of error without restar

Unhandled exceptions restart the device. The solution is to handle the exception by surrounding the call with a `try/catch` block. But, "in use" generally means you are trying to allocate some IO resource more than once, which would be a bug , so maybe better to fix the error than suppress the exception.

mauroForlimpopoli, 2025-03-27:
Good morning Peter, 
"to fix the error than suppress the exception" is absolutely right, but sometimes the error is not from code. (I think)
In my example, I do a 
startScanning(); 
and immediately after the discovery of the right device
stopScanning();  and then 
this.connect(device);.
This works correctly 99 times on 100, but sometimes it fails and the device restart.
I find a solution for this problem

phoddie, 2025-03-27:
> ...but sometimes the error is not from code. (I think)

It would be surprising for the device to spontaneously reset without any indication to the software.

> This works correctly 99 times on 100, but sometimes it fails and the device restart.

Try running an instrumented build (pass `-i` to `mcconfig` instead of `d`)..  If it the software is notified in anyway of the failure, the ESP-IDF should log that to the console, perhaps with a stack trace.  Perhaps that will provide you a clue.

mauroForlimpopoli, 2025-03-31:
.... If I understood well I done:
mcconfig -i -m -r 90 -p esp32/moddable_two
but I got
### Error: No such file or directory.
I add that I'm using moddable two with IDF 4.2

phoddie, 2025-03-31:
I don't know what's going on. There's no difference in environment dependencies between the `-d` and `-I`. If building with `-d` works for you, using `-I` instead should work.

> I add that I'm using moddable two with IDF 4.2

I assume that also means you are using an older Moddable SDK, as IDF 4.2 isn't supported with the current Moddable SDK. If that is indeed the case, it is an essential detail that was omitted from the original report.

mauroForlimpopoli, 2025-04-01:
yes, for some reasons i need to use old Moddable SDK (I don't know the release) . The xsbug if from 2017