Moddable-OpenSource/moddable #1472: Tap handlers not firing on behavior

danfinlay, 2025-03-19:
Hi, I got a moddable six and have been trying to build a very simple application, but for some reason when I write my own code, touch handlers on Behaviors aren't firing!

In the below code, the behavior's onCreate _does_ get triggered, but none of the touch handlers do. I've tried lots of combinations of different kinds of containers, with active/not active on the parents and so forth.

I read over the docs, and am left with some open questions:
- What is the full list of Behavior hooks?
- What is the way that `active` is used? Do all parents need to be `active` for a child to be?

Otherwise, I'm quite stumped. The examples all work fine, but this script does not:
(The script should work fine if you paste it over the piu/drag example app).
```
/*
 * Copyright (c) 2016-2020 Moddable Tech, Inc.
 *
 *   This file is part of the Moddable SDK.
 *
 *   This work is licensed under the
 *       Creative Commons Attribution 4.0 International License.
 *   To view a copy of this license, visit
 *       <http://creativecommons.org/licenses/by/4.0>.
 *   or send a letter to Creative Commons, PO Box 1866,
 *   Mountain View, CA 94042, USA.
 *
 */

import {} from "piu/MC";

const blackSkin = new Skin({ fill:"black" });
const whiteSkin = new Skin({ fill:"white" })
const blueSkin = new Skin({ fill:"#CCE7FF" });
const darkBlueSkin = new Skin({ fill:"#89B0FF" });
const textStyle = new Style({ font:"myFont", left:10, right:10, top:15, bottom:15 });
const centerStyle = new Style({ horizontal:"center" });
const leftStyle = new Style({ horizontal:"left" });
const rightStyle = new Style({ horizontal:"right" });
const justifyStyle = new Style({ horizontal:"justify" });
const absoluteLeadingStyle = new Style({ horizontal:"left", leading:30 });
const relativeLeadingStyle = new Style({ horizontal:"left", leading:-80 });
const otherStyle = new Style({ font:"OpenSans-Semibold-28", color:["blue","red"] });
const redStyle = new Style({ color:["red","blue"] });

let MyApplication = Application(null, {
	skin:whiteSkin, style:textStyle,
	active: true,
	Behavior: class extends Behavior {
		onDisplaying(application) {
			if (application.height != 240 || application.width != 320)
				trace("WARNING: This application was designed to run on a 320x240 screen.\n");
		}
	}
});

class ButtonBehavior extends Behavior {
	onCreate(content) {
		trace("Button created\n");
	}
	onTouchBegan(content, id, x, y, ticks) {
		trace("Touch began\n");
	}
	onTouchEnded(content, id, x, y, ticks) {
		trace("Touch ended\n");
	}
	onTouchMoved(content, id, x, y, ticks) {
		trace("Touch moved\n");
	}
}

const appTitle = Text(null, {
	left:0, right:0, top:0, skin:whiteSkin, style:centerStyle, active:true,
	blocks: [
		{ spans: [
			{ style:redStyle, spans: "Header" },
		]},
	],
});

const logo = new Content(null, {
	top: 0, height: 92, width: 100,
	style: centerStyle,
	skin: new Skin({ texture: new Texture("logo_100.png"), fill: "white", width: 100, height: 92 }),
});

const LabeledButton = Container.template($ => ({
	left:16, right:16, top:8, height: 44, skin:blueSkin, style:centerStyle, active:true,
	Behavior: ButtonBehavior,
	contents: [
		Label(null, {
			string: $.label,
			style: textStyle,
			active: true,
			Behavior: ButtonBehavior,
		})
	],
}));

const HomeScreen = new Column(null, {
	top: 0, bottom: 0, left: 0, right: 0,
	active: true,
});

HomeScreen.add(appTitle);
HomeScreen.add(logo);
HomeScreen.add(new LabeledButton({ label: "Connect" }));
HomeScreen.add(new LabeledButton({ label: "Inventory" }));
MyApplication.add(HomeScreen);

export default MyApplication;
```

phoddie, 2025-03-19:
As posted, it doesn't quite run for me in the piu/drag example.. I had to replace `"myFont"` with `"OpenSans-Semibold-28"`. With that, it runs in the simulator and the touch events are traced:

```
mcconfig -d -m -p sim/moddable_six
```

I'll try it later on Moddable Six hardware.

FWIW – `LabeledButton` has `Behavior: ButtonBehavior` at two points in the hierarchy. One should be enough. But, I don't expect that's the cause of your missing events.

> What is the full list of Behavior hooks?

The events vary by object, and you can [add your own](https://www.moddable.com/documentation/piu/piu#content-functions). Consequently, there is no definitive single list of events that a Behavior may receive. The Piu documentation lists the events associated with each object. For example, the [events](https://www.moddable.com/documentation/piu/piu#content-events) that `Content` object generates.

danfinlay, 2025-03-20:
Thank you so much for taking the time to try on both the simulator and the device.

I committed a cardinal sin here and assumed that because I had copied off that example (and it had worked at first) that it was still a good simulacra of the file's environment.

I made a repository that fails the way my local one does here:
https://github.com/danfinlay/moddable-repro

Locally, this repository works for me in the simulator but not on the hardware.

I'm also going to try to start from scratch and see if I can't isolate the problem by trying to cherry pick my changes so far.

phoddie, 2025-03-20:
The repository link above is failing with a 404. Is it set to private?

danfinlay, 2025-03-20:
Ah! It was! Thank you for your patience..

danfinlay, 2025-03-20:
So strange, now I'm getting different results again.

Formerly:
- I would get the `created` logs, but no more on the real device (simulator was working).

Now:
- I'm not getting logs of ANY sort
- I AM actually able to register behavioral handlers that I can observe the results to (so it's working actually on device!)
- However switching back to my original project it ISN'T working now.

I'm afraid I just don't even have a real repro, and my issues seem sporadic and inconsistent now.

Is there something I can do to re-connect xsbug to show logs? Somehow it seems like it isn't reflecting the current connection anymore.

phoddie, 2025-03-20:
> On my main project on the simulator, I get Error: display list overflowed (in onIdle)!.

I'm not sure what the "main project" is. The one I ran just has a few objects and shouldn't come close to overflowing the display list. But, if you draw many objects in a single frame, eventually you might need to use a bigger display list. Here's an example pulled from one of our projects:

```js
	globalThis.application = new Application(
		{ 
			model: new SystemModel,
		}, 
		{
			skin: assets.skins.black,
			commandListLength: 4096,
			displayListLength: 4096,
			behavior: controller
		}
	);
```

> 'm not getting logs of ANY sort

This isn't a Moddable SDK software problem or a Moddable Six hardware. Logging works. Really. ;) 

Two directions for you:

- Are you using USB (building with -`p esp32/moddable_six_cdc` or Serial  (`-p esp32/moddable_six`) to connect?
- Try some simple project like $MODDABLE/examples/base/timers`. Does that work reliably, trace as expected, etc?

danfinlay, 2025-03-20:
Oh I'm sure you're right and this isn't a moddable issue, I'm just not sure what I'm doing wrong!

I was using serial, apparently. I wasn't aware of the _cdc variant. I am now trying that.

Trying timers, it works. Even opens xsbug and logs to it correctly.

Building the repro example I gave you without CDC:
- Behaviorally works, but doesn't attach to xsbug or log to it anymore.

Building the repro example I gave you with CDC:
- Attached to XSbug and worked!

So it seems I was having serial problems, and USB largely fixed them.

That's great!... I think that means we got me through this one very silly feeling issue! Thank you!

I hope to have more issues again soon , and if I get time to keep hacking on this project. Thank you!!

phoddie, 2025-03-20:
FWIW – the [Building and Deploy Apps](https://www.moddable.com/documentation/devices/moddable-six#building-and-deploying-apps) section of the Moddable Six documentation has notes and photos about both USB and Serial.

danfinlay, 2025-03-20:
Thanks, I'll read that.

I seem to have hit my first "stack too deep" issues. Running some cryptography like this:

```
import { addr } from 'micro-eth-signer';
const random = addr.random(); // Secure: uses CSPRNG
const address = random.address;
```

Would it sound right that an esp32 runs out of memory trying to perform some cryptographic functions like that?

phoddie, 2025-03-20:
> Would it sound right that an esp32 runs out of memory trying to perform some cryptographic functions like that?

Dunno – not familiar with CSPRNG. But the JavaScript stack isn't the native stack. The default JavaScript stack isn't that big - 8 KB (512 slots on a 32-bit architecture). Actually, that's big enough for just about everything we usually do. But, you can bump that. At the top level of your `manifest.json`, do:

```
"creation": {
    "stack": 1024
}
```

More details in the [manifest docs](https://www.moddable.com/documentation/tools/manifest#creation).

danfinlay, 2025-03-21:
Expanded the stack but now just got memory full!

And this was basically the most minimal pure JS implementation of an ethereum signer I could find.

If I want much more memory than this I'll probably need a raspberry pi express or something, you think?

Know of any products like the moddable six with that?

danfinlay, 2025-03-21:
I see some of these include display guides, I suppose that's a good start: https://www.moddable.com/documentation/devices/pico

phoddie, 2025-03-21:
You give up far too easily. ;)

Is there a project I can try so I can look at what's going on?

danfinlay, 2025-03-21:
I'm keeping the repository private for an internal hackathon, but I've sent you an invite to it, should be in your email..

phoddie, 2025-03-21:
Thanks. I can see the repo now. Will take a look.