Moddable-OpenSource/moddable #1481: Config key name mismatch for Hume AI

stc1988, 2025-04-05:
There are two key names `humeAIKey`  and `humeAPIKey` for API keu of Hume AI.
Based on the [documentation](https://dev.hume.ai/docs/text-to-speech-tts/quickstart/typescript), `humeAPIKey` seems to be better.

phoddie, 2025-04-06:
Oops. The documentation uses `humeAPIKey` and the code uses `humeAIKey`.

There is precedent for both: we have `openAIKey` and `geminiAPIKey`.

"openAI" and "humeAI" are names of the company / service, so maybe that is consistent. The inconsistency is "geminiAPIKey".  Also, correcting the documentation to match the code won't break anything. Maybe that's OK?

stc1988, 2025-04-06:
> Also, correcting the documentation to match the code won't break anything. Maybe that's OK?

Of course, no problem.

closed 2025-05-26 by stc1988