Moddable-OpenSource/moddable #1489: Include error while building nrf52 on Windows.

NW-Lab, 2025-04-26:
hello

When building nrf52 on Windows, I get an error saying that the mc.xs.h file is missing.

Build environment:Windows 10
Moddable SDK version:Current repository(5.6.0)
Target device:nrf52/xiao

[image]

Thank you

phoddie, 2025-04-26:
The `mc.xs.c` file is generated by the build. This error is likely the consequence of a missing dependency in the nmake file for nRF52. We'll take a look.

phoddie, 2025-04-26:
We are unable to reproduce the same result here on Windows. Checking the `nmake.nrf52.mk` show that xsmain.c.o depends on the existence of `mc.xs.c`. The build step that creates `mc.xs.c` using `$(XSL)` also creates `mc.xs.h`:

```make
$(TMP_DIR)\mc.xs.c: $(MODULES) $(MANIFEST)
	@echo # xsl modules
	$(XSL) -b $(MODULES_DIR) -o $(TMP_DIR) $(PRELOADS) $(STRIPS) $(CREATION) -u / $(MODULES)
...
$(TMP_DIR)\xsmain.c.o: $(BUILD_DIR)\devices\nrf52\base\xsmain.c $(TMP_DIR)\mc.xs.c
	@echo # application: $(@F)
	$(CC) $(C_FLAGS) $(C_DEFINES) $(C_INCLUDES) $? -o $@
```

The build output you shared appears to be from an incremental build. It is possible than an earlier error left the output in an unexpected state. Would you try a clean build? If that still fails, please provide the full output. Also, please check to see if `mc.xs.c` exists in the output directory. To do a clean build:

```
mcconfig -d -m -p nrf52/xiao -t clean
mcconfig -d -m -p nrf52/xiao
```

Thank you.

NW-Lab, 2025-04-27:
Thank you for your reply.

[image]

It's been a while since I last used nrf52, so I may have placed the SDK incorrectly.
Since you can't reproduce the problem, I'll check these installation instructions for mistakes.

Thank you.

NW-Lab, 2025-04-27:
I've tried various things, but I think namke.nrf.mk is wrong.
But I just don't know how to fix it.

nmake file(line:474)
```
SDK_GLUE_OBJ = \
	$(TMP_DIR)\debugger.o \
	$(TMP_DIR)\ftdi_trace.o \
	$(TMP_DIR)\main.o \
	$(TMP_DIR)\systemclock.o \
	$(TMP_DIR)\xsmain.o \
```
Line:990 is xmain.c.o
```
$(TMP_DIR)\xsmain.c.o: $(BUILD_DIR)\devices\nrf52\base\xsmain.c $(TMP_DIR)\mc.xs.c
	@echo # application: $(@F)
	@echo   $(TMP_DIR)
	$(CC) $(C_FLAGS) $(C_DEFINES) $(C_INCLUDES) $? -o $@
```

An error occurs, but if you change line 990 to xmain.o and build it, then change it back and build it, the error disappears.

Thank you

phoddie, 2025-04-28:
@NW-Lab, thank you for your investigation.

It looks like the problem is a bad dependency. On Windows the object files are named without `.c` so `xsmain.o` whereas on macOS and Linux they are `xsmain.c.o`. That means line 990 above is ignored. This error probably only happens on certain machines because the degree of parallelism in the build depends on the capabilities of the computer.

The solution is what you did: change `xsmain.c.o` to `xsmain.o`. I've committed that as part of the next public Moddable SDK release.

Thank you!

NW-Lab, 2025-04-29:
I understand the cause of the error.

If I just change xsmain.c.o to xmain.o here, I get the error "arm-none-eabi-gcc.EXE: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files".
'''
c:\pjt\moddable\examples\helloworld>mcconfig -d -m  -p nrf52/xiao
# application: debugger.o
# application: ftdi_trace.o
# application: main.o
# application: systemclock.o
# xsc Resource.xsb
# xsc preference.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc main.xsb
# xsc pins\digital.xsb
# xsc pins\digital\monitor.xsb
# xsc button.xsb
# xsc led.xsb
# xsc setup\target.xsb
# xsc mc\config.xsb
# xsid Resource.c.xsi
# xsid modPreference.c.xsi
# xsid modGPIO.c.xsi
# xsid modMonitor.c.xsi
# xsid modTime.c.xsi
# xsid modTimer.c.xsi
# xsid timer.c.xsi
# xsid modInstrumentation.c.xsi
# xsid digital.c.xsi
# xsid modPreference.h.xsi
# xsid modGPIO.h.xsi
# xsid modTimer.h.xsi
# xsid modInstrumentation.h.xsi
# xsl modules
### 299 instances, 597 keys, 65 colors, 0 holes
# application: xsmain.o
arm-none-eabi-gcc.EXE: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
NMAKE : fatal error U1077: 'C:\Users\tauchi\nrf5\arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi\bin\arm-none-eabi-gcc -c  -std=gnu99  --sysroot=C:\Users\tauchi\nrf5\arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi\arm-none-eabi  -fdata-sections  -ffunction-sections  -fmessage-length=0  -fno-builtin  -fno-common  -fno-diagnostics-show-caret  -fno-dwarf2-cfi-asm  -fno-strict-aliasing  -fomit-frame-pointer  -fshort-enums  -gdwarf-3  -gpubnames  -mcpu=cortex-m4  -mfloat-abi=hard  -mlittle-endian  -mfpu=fpv4-sp-d16  -mthumb   -mthumb-interwork  -mtp=soft  -munaligned-access  -nostdinc  -D__SIZEOF_WCHAR_T=4  -D__ARM_ARCH_7EM__  -D__ARM_ARCH_FPV4_SP_D16__  -D__HEAP_SIZE__=0x35000  -D__GNU_LINKER  -DBOARD_MODDABLE_FOUR  -DCONFIG_GPIO_AS_PINRESET  -DFLOAT_ABI_HARD  -DFREERTOS  -DINCLUDE_vTaskSuspend  -DINITIALIZE_USER_SECTIONS  -DNO_VTOR_CONFIG   -DNRF52840_XXAA  -DNRF_SD_BLE_API_VERSION=7  -DNRF_USBD_REQUIRE_CLOSED_ON_PORT_OPEN=1  -DS140  -DSOFTDEVICE_PRESENT  -Dnrf52  -DAPP_TIMER_V2  -DAPP_TIMER_V2_RTC1_ENABLED  -DMBEDTLS_CONFIG_FILE=\"nrf_crypto_mbedtls_config.h\"  -DNRF_CRYPTO_MAX_INSTANCE_COUNT=1  -DSVC_INTERFACE_CALL_AS_NORMAL_FUNCTION    -DmxUseDefaultSharedChunks=1  -DkCommodettoBitmapFormat=7  -DkPocoRotation=0  -DMODGCC=1  -DUSE_FTDI_TRACE=0 -DMODINSTRUMENTATION=1 -DmxInstrument=1 -DmxDebug=1 -DDEBUG=1 -DUSE_DEBUGGER_USBD=1 -g2 -Os -iprefix C:\Users\tauchi\nrf5\arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi\  -iwithprefix arm-none-eabi\include  -iwithprefix arm-none-eabi\include\machine  -iwithprefix lib\gcc\arm-none-eabi\12.2.1\include  -iwithprefix lib\gcc\arm-none-eabi\12.2.1\include-fixed   -Ic:\pjt\moddable\modules\files\preference -Ic:\pjt\moddable\modules\pins\digital\nrf52 -Ic:\pjt\moddable\modules\base\timer -Ic:\pjt\moddable\modules\base\instrumentation -Ic:\pjt\moddable\build\devices\nrf52  -Ic:\pjt\moddable\build\devices\nrf52\config -iprefix c:\pjt\moddable\xs\  -iwithprefix includes  -iwithprefix sources  -iwithprefix platforms\mc  -iwithprefix platforms\nrf52  -iwithprefix ..\modules\files\preference  -iwithprefix ..\modules\base\instrumentation  -iwithprefix ..\modules\base\timer  -iprefix c:\pjt\moddable\build\devices\nrf52\  -iwithprefix base  -iwithprefix config  -iwithprefix xsProj -Ic:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib -Ic:\pjt\moddable\build\tmp\nrf52\xiao\debug\helloworld -Ic:\pjt\moddable\build\devices\nrf52 -iprefix C:\Users\tauchi\nrf5\nRF5_SDK_17.0.2_d674dde\components\libraries\  -iwithprefix crypto  -iwithprefix crypto\backend\cc310  -iwithprefix crypto\backend\cc310_bl  -iwithprefix crypto\backend\cifra  -iwithprefix crypto\backend\nrf_hw  -iwithprefix crypto\backend\mbedtls  -iwithprefix crypto\backend\micro_ecc  -iwithprefix crypto\backend\nrf_sw  -iwithprefix crypto\backend\oberon  -iwithprefix crypto\backend\optiga  -iwithprefix atomic  -iwithprefix atomic_fifo  -iwithprefix atomic_flags  -iwithprefix balloc  -iwithprefix button  -iwithprefix bsp  -iwithprefix delay  -iwithprefix experimental_section_vars  -iwithprefix fds  -iwithprefix fifo  -iwithprefix fstorage  -iwithprefix hardfault  -iwithprefix hardfault\nrf52  -iwithprefix hardfault\nrf52\handler  -iwithprefix libuarte  -iwithprefix log  -iwithprefix log\src  -iwithprefix memobj  -iwithprefix mutex  -iwithprefix queue  -iwithprefix ringbuf  -iwithprefix scheduler  -iwithprefix serial  -iwithprefix sortlist  -iwithprefix stack_info  -iwithprefix strerror  -iwithprefix twi_sensor  -iwithprefix twi_mngr  -iwithprefix timer  -iwithprefix util  -iwithprefix usbd  -iwithprefix usbd\class\cdc  -iwithprefix usbd\class\cdc\acm  -iprefix C:\Users\tauchi\nrf5\nRF5_SDK_17.0.2_d674dde\  -iwithprefix external\nrf_cc310\include  -iwithprefix external\nrf_cc310_bl\include  -iwithprefix external\freertos\portable\GCC\nrf52  -iwithprefix external\freertos\portable\CMSIS\nrf52  -iwithprefix external\freertos\source  -iwithprefix external\freertos\source\include  -iwithprefix external\freertos\source\portable\MemMang  -iwithprefix components  -iwithprefix components\ble\common  -iwithprefix components\ble\ble_advertising  -iwithprefix components\ble\ble_radio_notification  -iwithprefix components\ble\nrf_ble_gatt  -iwithprefix components\ble\nrf_ble_qwr  -iwithprefix components\ble\nrf_ble_scan  -iwithprefix components\ble\peer_manager  -iwithprefix components\boards  -iwithprefix components\softd [... cut ...]  -iwithprefix modules\nrfx  -iwithprefix modules\nrfx\drivers\include  -iwithprefix modules\nrfx\drivers\src  -iwithprefix modules\nrfx\drivers\src\prs  -iwithprefix modules\nrfx\hal  -iwithprefix modules\nrfx\mdk  -iwithprefix modules\nrfx\soc c:\pjt\moddable\build\devices\nrf52\base\xsmain.c c:\pjt\moddable\build\tmp\nrf52\xiao\debug\helloworld\mc.xs.c -o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\helloworld\xsmain.o' : リターン コード '0x1'
Stop.

c:\pjt\moddable\examples\helloworld>
'''

Thank you.

phoddie, 2025-05-05:
Just closing the loop here, @mkellner solved the above problem with `nmake` builds for nRF52. That will be in our forthcoming May release. Please verify that when it is available. Thank you.

closed 2026-05-21 by mkellner