Moddable-OpenSource/moddable #1490: clearTimeout crashes

ChALkeR, 2025-05-02:
**Build environment:** macOS
**Moddable SDK version:**
```console
~/.jsvu/bin/xs -v
XS 16.8.1, slot 32 bytes, ID 4 bytes
```

**Description**
`clearTimeout` crashes:
1. on `undefined`
2. when called on the timeout id from that timeout callback
    In this case not `clearTimeout()` crashes, but internal handler -- see example 3

Case 2 is not catchable.

**Steps to Reproduce**
1. `~/.jsvu/bin/xs -e 'clearTimeout(undefined)'`
2. `~/.jsvu/bin/xs -e 'x=setTimeout(()=>clearTimeout(x), 100)'`
3. `~/.jsvu/bin/xs -e 'x=setTimeout(()=>{clearTimeout(x);print("cleared")}, 100)'`

**Expected behavior**
It should do nothing if there is no timeout to remove

https://www.w3.org/TR/2011/WD-html5-20110525/timers.html#dom-windowtimers-cleartimeout
> If handle does not identify an entry in the [list of active timeouts](https://www.w3.org/TR/2011/WD-html5-20110525/timers.html#list-of-active-timeouts) of the [WindowTimers](https://www.w3.org/TR/2011/WD-html5-20110525/timers.html#windowtimers) object on which the method was invoked, the method does nothing.

phoddie, 2025-05-02:
Thank you for the detailed report. The problem is easily reproduced. I've fixed this for our next release (likely in the next week). If you'd like to try it sooner, I'll post the patch – just let me know.

ChALkeR, 2025-05-20:
Thanks! I can confirm this is fixed in 5.7.0, tests pass now in https://github.com/ExodusMovement/test/commit/9b88d64089c0b5875a4dbcc652e3cf9e969072f4.

Side not: jsvu needs updating in https://github.com/GoogleChromeLabs/jsvu/blob/main/engines/xs/predict-url.js, as in 5.7.0 the name of the macOS build changed.

closed 2025-05-20 by ChALkeR

phoddie, 2025-05-20:
We'll take a look at updating jsvu. We've mostly been using esvu.

ChALkeR, 2025-05-20:
@phoddie `esvu` likewise fails on 5.7.0

phoddie, 2025-05-21:
We landed a [PR](https://github.com/GoogleChromeLabs/jsvu/pull/136) to jsvu some time ago. Is it possible your local copy is out of date?

ChALkeR, 2025-05-21:
@phoddie No.

jsvu:
https://github.com/GoogleChromeLabs/jsvu/blob/4968c3c0d562695bf2594d1c4a6a10f99ba47c3d/engines/xs/predict-url.js#L18-L23

esvu:
https://github.com/devsnek/esvu/blob/09a20f30533dbc4c3f3d6c5165f478fe4c998dff/src/engines/xs.js#L11-L14

[5.6.0](https://github.com/Moddable-OpenSource/moddable/releases/tag/5.6.0):
[image]

[5.7.0](https://github.com/Moddable-OpenSource/moddable/releases/tag/5.7.0):
[image]

phoddie, 2025-05-23:
I think you are pointing out that the Mac binaries are incomplete in 5.7.0 but were correct in 5.6.0. That being the case, it isn't esvu and jsvu that need updating, but the release.

ChALkeR, 2025-05-23:
@phoddie `macuniversal` builds work

I assumed not packing `mac64`/`mac64arm` in favor of `macuniversal` was deliberate -- in that case, both esvu and jsvu need to be updated to fetch `macuniversal` for versions from 5.7.0 onwards

If that was accidental, then yes, missing builds need to be uploaded

phoddie, 2025-05-23:
Yeah, understood. You were perfectly clear in the report, but I got led down the path of assuming it was esvu/jsvu that need updating when actually it is our release. I think was just a mistake in our release process this month. We'll get it sorted out.

phoddie, 2025-05-23:
The 5.7.0 binaries release has been updated. I think this looks right now:

[image]

Thanks again.