Moddable-OpenSource/moddable #1492: mcpack is failed with esp32 device

stc1988, 2025-05-03:
**Build environment:** macOS
**Moddable SDK version:**  5.6.0
**Target device:** NodeMCU ESP32

**Steps to Reproduce**
```
cd $MODDABLE/examples/package/hello
npm install
mcpack mcconfig -d -m -p esp32/nodemcu

/Users/satoshi/Projects/moddable/modules/network/ethernet/manifest.json:1: error: '$(ETHERNETDRIVER)/*': directory not found!!
### 1 error(s)
```

phoddie, 2025-05-03:
The `esp32/nodemcu` target doesn't include $MODDABLE/modules/network/ethernet/manifest.json and neither does $MODDABLE/examples/package/hello/package.json. Where is that coming from?

stc1988, 2025-05-03:
> The esp32/nodemcu target doesn't include $MODDABLE/modules/network/ethernet/manifest.json and neither does $MODDABLE/examples/package/hello/package.json. Where is that coming from?

I think the same way. Can you reproduce it on your end?

phoddie, 2025-05-03:
`mcpack` scans $MODDABLE/modules` to find the mappings from module to manifest.  That allows it to automatically construct the manifest includes from the `import` statements in your source code. Unfortunately, Espressif moved the Ethernet ENC28J60 driver from the IDF to the Component Registry. That broke the path causing the scan to fail and, consequently, halting the build.

The quick fix is to update the Ethernet manifest to [use the Component Registry](https://moddable.com/blog/using-the-esp-component-registry/). That allows the build to succeed. Unfortunately, other Espressif changes broke the Ethernet module build. That looks fixable but I'm not doing that today. Here's a patch to the ESP32 section of the `platforms` section of `modules/network/ethernet/manifest.json` that gets the build working:

```json
        "esp32": {
            "build": {
                "ETHERNETDRIVER": "$(IDF_PATH)/examples/ethernet/enc28j60/components/eth_enc28j60"
            },
			"dependency": [
				{
					"name": "enc28j60",
					"version": "^1.0.1",
					"includes": [
						"include"
					]
				}
			],
            "include": [
                "$(MODULES)/pins/spi/manifest.json"
            ],
            "modules": { 
                "*": [
                    "$(ETHERNET)/*",
                    "$(ETHERNET)/esp32/*",
                    "$(ETHERNET)/esp32/drivers/enc28j60/*"
                ]
            }
```

stc1988, 2025-05-03:
Thank you for confirming. The build started working with the method you suggested.

closed 2025-05-26 by stc1988