Moddable-OpenSource/moddable #1497: Touch not works on M5Stack CoreS3

stc1988, 2025-05-11:
**Build environment:** macOS
**Moddable SDK version:** 5.6.0
**Target device:** m5Stack Cores3

**Steps to Reproduce**
1. Build and install the `examples/commodetto/mini-drag` app using this build command: `mcconfig -d -m -p m5stack_cores3`

The following message is displayed in xsbug.

```
/Users/satoshi/Projects/moddable/build/devices/esp32/targets/m5stack_cores3/M5StackCoreS3Touch.js (42) # Break: TypeError: call: not a function (in M5StackCoreTouch.prototype.read)!
```

**Other information**

In [fabf5e6](https://github.com/Moddable-OpenSource/moddable/commit/fabf5e66856717585099a976bbc26632922626cf), Touch driver is replaced to Ecma419, but It seems the implementation also needs to be changed accordingly.

phoddie, 2025-05-13:
The example expects the original touch driver, as you noted. Options:

- Check for the new driver and fail with an error message
- Update the example to support both
- Migrate the example to the new API only (error message on original)

I'm not sure the complexity of supporting both is justified here. The goal is to migrate everything to the 419 APIs so work on backwards compatibility is not so valuable. Maybe an error message is enough for now. What do you think?

stc1988, 2025-05-17:
Sorry, I was indeed testing with an example that doesn’t work.

> The goal is to migrate everything to the 419 APIs so work on backwards compatibility is not so valuable. Maybe an error message is enough for now. What do you think?

Yes, I also think an error message is enough.

I also try to run `examples/piu/drag` , but  I got an error `/Users/satoshi/Projects/moddable/modules/io/i2c/esp32/_i2c.c (367) # Break: Error: writeRead failed (in writeRead)!`. Is this due to a recent update?

phoddie, 2025-05-20:
For the `writeRead` exception, it seems related to the new I²C stack from Espressif. There is a mystery flag that seems to help. In this structure....

https://github.com/Moddable-OpenSource/moddable/blob/2e7111b3d1f7abd907b9716560b6628b7c4d2c91/modules/io/i2c/esp32/_i2c.c#L187-L191

.... try adding:

```c
		.flags.disable_ack_check = 1,
```

Please let me know if that makes a difference.

stc1988, 2025-05-21:
I tried it, but the situation didn’t change.

stc1988, 2025-06-12:
The issue was resolved by reverting Espressif’s new i2c_master.