Moddable-OpenSource/moddable #1500: BLE connect on address

mauroForlimpopoli, 2025-05-16:
Good afternnon,
I'm using Moddable Six with IDF 5.3.0
To connect to a device I use the "quick" connect command  inside the 
"client = new UARTClient;"
```
let buffer = new ArrayBuffer(6); 
let view = new Uint8Array(buffer); 
view[0] = parseInt(bleName.slice(15,17), 16);
view[1] = parseInt(bleName.slice(12,14), 16);
view[2] = parseInt(bleName.slice(9,11), 16);
view[3] = parseInt(bleName.slice(6,8), 16);
view[4] = parseInt(bleName.slice(3,5), 16);
view[5] = parseInt(bleName.slice(0,2), 16);
this.connect({
	address: buffer,
	addressType: 0
});
``` 
If the BLE device is presene everithing works but if the device is not present I need to close and re-open the session arter 5 seconds with

```
if(client != null){
	client.close()
	client = null
}
.....
this.connect({
	address: buffer,
	addressType: 0
});
.....
client = new UARTClient;
``` 
and if the device is not present again when I repeat the session
```
	if(client != null){   <----- this is false
		client.close()
		client = null
	}
 client = new UARTClient;    <--- 
```
I get the error 
....Projects\moddable\modules\network\ble\nimble\modBLEClient.c (236) # Break: BLE already initialized (in initialize)!

In a few words it seems that the command 
```
this.connect({
	address: buffer,
	addressType: 0
});
```
can be close only once. the second cause the error above. If a connection is done I can close and re-open again
Is it possible?
regards

phoddie, 2025-05-21:
When closing the client, it waits for all transactions to complete to cleanly terminate. This is asynchronous. But, connecting to a non-existent device never completes - NimBLE waits forever for the device to appear. So, the client never actually closes. That leads to your exception when attempting to create a new client.

You can try canceling the connection attempt when closing. In `$MODDABLE/modules/network/ble/nimble/modBLEClient.c` inside `xs_ble_client_close` add a call to `ble_gap_conn_cancel()`.

mauroForlimpopoli, 2025-05-26:
Peter,
I tried your solution but no good result. 
Have you some other ideas?
Furthermore, sometimes I get this error:
C:\moddable\modules\network\ble\nimble\modBLEClient.c (418) # Break: (host): connection not found!
What exactly means? I can I avoid it?
regards

closed 2026-04-21 by phoddie