Moddable-OpenSource/moddable #1506: Cannot get mcrun to work for ch1_example Helloworld

rmontrosecbw, 2025-06-10:
Using Moddable 5.7.0 (moddable\tools\VERSION) on Windows 11, Espressif tools 5.4, Moddable six board. I can compile the examples with mcconfig and they load and run as expected. I'm trying to get the Mods environment running so we can compile just the Javascript code and have a shorter build cycle.

I'm trying to use the chapter 1 Helloworld example from the book. I build the host environment **mcconfig -d -m -p esp32/moddable_six_cdc** to compile and load into the Moddable Six using the USB JTAG port. XsBug shows
```
No Wi-Fi SSID
Device flashed. Ready to install apps.
```
as expected.

I then followed the book and run **mcrun -d -m -p esp32/moddable_six_cdc**, and that appears to work. I see
```
Starting xsbug...
"Running serial2xsbug..."
```
and XSBug comes up, but I never see the **Hello World** message in the XSBug log. 

If I attach a terminal program to COM3 I see:
```
<?xs-00000000?>

<?xs.3FCB4DA0?>
 <xsbug><login name="main" value="XS 16.5.0 32-bit LE 16-bit ID" flags="e"/></xsbug>

<?xs-3FCB4DA0?>
```

So it appears the Host program is loaded onto the Moddable Six.  

I have also tried this with the Moddable Six USB adapter (changing the platform to esp32/moddable_six) and got the same results. I have erased the Flash before running these steps with no change.

Looking through the Github discussions and Glitter I see where people have errors running mcrun, but not where there are no errors but it doesn't appear to be running.

Do you have any advice on where to look? We're new to Moddable and want to use it on a project.  Here's a capture of the host and example build 

[mcrun_build.txt](https://github.com/user-attachments/files/20679958/mcrun_build.txt)

phoddie, 2025-06-11:
I don't have access to a Windows machine at the moment. A couple observations:

- This works just fine on macOS. The host installed by mcconfig and the mod generated by mcrun should both be fine.
- From your report, it looks like the mod installed correctly. The problem is that the device isn't connecting to xsbug.

Two things you can try:

1. As a quick experiment, after `mcrun` hangs up, hit Control C, and then try re-connecting to xsbug with `mcconfig`, since that worked for you: `mcconfig -d -m -p esp32/moddable_six_cdc -t xsbug`.
2. It looks like `mcrun` on Windows relies on the `UPLOAD_PORT` environment variable to know the serial port the microcontroller is attached to. Try setting that to `COM3` and  re-run `mcrun`.

rmontrosecbw, 2025-06-12:
Thank-you for verifying that this should work. I finally got it working and can make Module builds to run on a device with a host environment.  I tried repeating the build after setting ``UPLOAD_PORT`` didn't change the outcome.

Here's what I did to fix things:

1. I cleaned and rebuilt all the tools in **%MODDABLE%\build\makefiles\win** from an ESP32 5.4 Command prompt that setup the vcvars32.bat. When I upgraded to 5.7 and rebuilt the tools, it was from a VC 2002 Command prompt, not the Espressif one.
(Note, on https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md it states you can have the command prompt setup with '**%comspec% /k ""%ProgramFiles%\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" && pushd %IDF_PATH% && "%IDF_TOOLS_PATH%\idf_cmd_init.bat" && popd"'**. Doing this, the Moddable build gives errors and does not build. You need to first load the Espressif settings AND then the moddable setting to get Moddable to build correctly (**%comspec% /k "pushd %IDF_PATH% && "%IDF_TOOLS_PATH%\idf_cmd_init.bat" && popd && "%ProgramFiles%\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat""**).
2. I then rebuilt the Host environment, which behaved as before.
3. I then built the Mod helloworld example using mcrun -d -m -p esp32/moddable_six_cdc. I got a very different build output. xsbug came up, then went into breakpoint mode and behaved as a standard xsbug tool. The build output is:

```
C:\Espressif5\tools\idf-exe\1.0.3\idf.py.exe
C:\Espressif5\frameworks\esp-idf-v5.4\tools\idf.py
# xsc example.xsb
# xsc check.xsb
"# xsl C:\CBW\Moddable\moddable\build\bin\esp32\debug\helloworld-gui2\helloworld-gui2.xsa"
Starting xsbug...
"Running serial2xsbug..."
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037c01e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1b0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xb84
load:0x403cb700,len:0x282c
entry 0x403c887c
Installing mod...complete
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403754b1
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1b0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xb84
load:0x403cb700,len:0x282c
entry 0x403c887c`
```

I modified the helloworld-gui to be a Module by moving the native code to the host environment and can load it as a Module using mcrun. 
The fact that the building was working using mcconfig threw me off.

closed 2025-06-12 by rmontrosecbw