Moddable-OpenSource/moddable #1514: DS18B20 setting

mauroForlimpopoli, 2025-07-07:
I'm using the example drivers/onewire/mail.js .

The instructions
```
  sensor.resolution = 9;
  res = sensor.resolution;
  trace("res - should be 9 :", res, '\n');
```
must configure the resoluton of the sensor and the example works but only during the test
If the device is not shunt down the setting is lost
In other words the configuration is done only in the RAM of the sensor and not in the  internal EEPROM.
Can you verify it? maybe is only a issue of my sensors, but I test a lot of them (from unique seller)
Thank you

mauroForlimpopoli, 2025-07-07:
...add  then I tried to reduce the pullup restistor until 1 kohm nothing is changed

mauroForlimpopoli, 2025-07-08:
I'm not using parassite power, but the sensor is supplied directly with 4 Vdc.
I also added a delay of 20 ms after command 
bus.write(0x48); // Write scratchpad
but nothing is changed.
I tried to program it with usb/1wire interface of Maxim and it works.
I add another function in DS18X20.js
```
  set writePad (a) {
	  
let scratchpad = this.scratchpad;
trace(scratchpad[2] , " scr2\n")
trace(scratchpad[3] , " scr3\n")
	  
    const bus = this.bus;
    bus.reset();
    bus.select(this.id);
	bus.write(0x48); // Write scratchpad
	Timer.delay(20);
    bus.reset();
  }
```
I call it after the instruction
sensor.resolution = 12;
.... but nothing changes

I get no error, but trace() instruction and "stop" debugger is not possibile in the DS18X20.js. Is it normal?

mauroForlimpopoli, 2025-07-11:
Hi Peter,
...."the argument a is unused"... is a forgetfulness of my previus code in that the delay time was variable.
No new  theory I just repeat the old sequence by adding a delay of 20 ms. Theoretically, by using direct supply, it is not necessary, but I tried to see if anything changed.
Can you tell me if the trace() instruction and "breakpoint debugger"  is possibile in the DS18X20.js ? because I don't get them

mauroForlimpopoli, 2025-07-17:
Good morning Peter,
the code if done sperimentally. I know that no reason on that
Using sensors with a long cable and until 4 sensor at the same time sometimes this is the code that given no issue.
Taking away the "Delay" and with only 1 "Reset" , sometimes the writing sequence fault. It could be is only attributable to the length of the cable and to the number of the sensors in the net, or maybe it's just a coincidence.
Anyway in this mode I had no issue. 
This is the result of my test
Thank you for your assistance

phoddie, 2025-07-17:
Thanks for explaining. The very long cable could certainly make a difference.

I've added the reset to the driver code, as that is required according to the data sheet.

closed 2025-07-17 by phoddie